Ultimaker/CuraEngine #1239: Feature request: rigidity-assistance support

richfelker, 2020-04-17:
This is particularly in regards to printing with TPU/flexible materials, but depending on geometry it can be an issue even for mostly-rigid ones.

It's possible to have models that don't "need support" whatsoever in the sense of not having nontrivial overhangs, but that nonetheless suffer from lack of rigidity due to tall and narrow components. With rigid materials these are subject to getting broken or causing the whole print to be knocked loose from the bed due to nozzle collisions after minor warping, and with flexible materials the nozzle tends to bend the part so much while moving that successive layers end up in the wrong place.

For an example where I hit this with TPU, see the tail in https://www.thingiverse.com/thing:3877926

Feature request is somewhat open-ended and not something I expect to happen right away: It would be nice if Cura could generate supports for rigidity-assistance only - (possibly-leaning) towers that periodically touch narrow components (component in the sense of a 2d-connected component in the slicing) to keep them in place for printing subsequent layers on top, but with minimal damage to the surface and difficulty in removal.

Ghostkeeper, 2020-04-22:
I've dealt with this myself as well. It's quite common when printing stuff with legs, like a unicorn or a stool or something. If I can't rotate the model to put the weak part at the top of the print, I usually fix it by reducing the overhang angle and then adding a few support blockers here and extra support cylinders there. Maybe increase the infill density at the weak point, too.

I'm not sure what would be a good approach to generate these automatically though. Testing rigidity of a model is really hard. The use case of that Pikachu model is a good example, with something big (which itself is pretty rigid) on top of something thin (which is not). To detect that, Cura would need to transfer forces across many layers. In other words, basically a do physical simulation. Now that's not totally out of the question, but still a far way off at least.

richfelker, 2020-04-22:
I think it could be done without any intensive physical simulation just by tracking, recursively, for each component of a layer, something like the minimal cross-sectional layer area (from previous layers) supporting it. If that's too small and the Z distance since the too-small layer is sufficiently large, add in a touching support tower. This support tower would then contribute to computing this metric for future layers, and delay or prevent the need for another one.

This would also help with situations like https://3dprinting.stackexchange.com/questions/13380/brim-around-just-one-section

Ghostkeeper, 2020-04-28:
I think that would also lead to a lot of false positives though. Most engineering-type models have very thin walls. For instance, a single-walled L-shape of 2x2cm is fairly sturdy and probably prints fine up to a height of like 10cm without vibrating, but it only has about `4mm x 0.4mm = 1.6mm²` cross-sectional area. But a circular pillar with 1.6mm² area would have a radius of 0.714mm, and I wouldn't trust this to print without vibrations for more than 1cm height or so.

Those are extreme examples of course, just to illustrate. But with engineering models you'll find lots of cases where area over height is not sufficient. With organic models you'll find lots of cases where something is not so thin normally but since it has a slight tilt, the overhang makes it weak again.

Mike111177, 2020-09-25:
The term that generally quantifies the stiffness based on cross section is second moment of area, which could be calculated for each individual polygon pretty easily. This accounts for thin walls as well. This value is also a directional vector so it would inform you of what direction it needs to be supported in. I can go into more detail if needed. You can also probably iterate on each layer to determine the amount of support it needs. So if you have a polygon on top of another that needs a bunch of support, consider adding support to this one as well. You can also consider if the polygon you are is suddenly connected to a section that was very rigid (say via a bridge) it inherits the stiffness of that sections it connects to.

https://en.wikipedia.org/wiki/Second_moment_of_area
https://en.wikipedia.org/wiki/List_of_second_moments_of_area