Ultimaker/CuraEngine #1240: [4.5.0] Inner Walls use twice as much paths than outer Walls resulting in printer slowdown

timoxd7, 2020-04-20:
**Application Version**
Cura 4.5.0 (Compiled version from Ultimaker Website)

**Platform**
Windows

**Qt**
Using Version from Ultimaker Website

**PyQt**
Using Version from Ultimaker Website

**Display Driver**
Nvidia GTX 1070 Mobile (Version 442.19)

**Steps to Reproduce**
Just generate GCODE from attached Project File

**Actual Results**
Outer Walls get printed fast (as expected). Inner Walls constantly slowing down and accelerating back to desired printing speed.
In the GCODE-File, inner walls use almost twice as many paths as the corresbonding outer walls.

**Expected results**
Printer printing with high speed at all parts and line types.
Inner Lines use almost the same amount of paths as the outer walls (should only be an inset of the outer wall).

**Additional Information**
Using 8-Bit Stock Ender 3 Board with modified Firmware. Printed from SD directly.
The Mesh is already optimized to match the printers processing speed. The Problem is, that the inner walls get processed with more detail than the outer walls, producing almost twice as much paths than the outer wall, resulting in a stuttering printer while it can't keep up with the GCODE commands in time. I also adjusted the mesh resolution and deviation in cura, but it did not changed that inner walls use almost twice as much paths than outer walls.

Uploaded Project File and GCODE (.zip): https://we.tl/t-tYJ2iKPgAE

smartavionics, 2020-04-20:
I looked at your gcode and if I understand the problem correctly I think the problem is due to the poor performance of the wall gap filling that UM Cura currently does. You can verify that by setting the fill gaps between walls setting to nowhere and then if you slice and all those inner line segments go away then you can be sure that it is the gap filling that is creating them.

I believe that UM are working on a solution but in the meantime you may wish to try my Cura builds which implement the gap filling differently. My releases can be installed alongside the UM releases and can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

Ghostkeeper, 2020-04-22:
It looks like your problem, as you describe it and as it's seen in the project file, is not really with the gap filling technique but rather with wall overlap compensation. Compensate Outer Wall Overlaps is disabled (and in this case wouldn't have much of an effect either) but Compensate Inner Wall Overlaps is enabled. Disabling that reduces the size of the g-code from 71.2MB to 54.5MB with your project. (I'm using a more modern version of Cura than you were when you created your 74.3MB file, so my initial result is slightly different too.)

The reason why this setting generates so much more g-code is because it checks for every line that the wall overlaps with, which part of the line overlaps with the other line. It separates that part out to be printed with a reduced flow rate to compensate for the overlap. Effectively this doubles the number of line segments for one of the two intersecting contours. Since your print consists almost exclusively out of long curves that follow each other very closely, this has a big effect on your CPU usage where otherwise your printer would catch up quickly after the curve.

When you disable this, you'll probably get some overextrusion. I think for the use case of these masks that probably won't do much harm if your extruder is precise and your nozzle flows consistently, although for the inside you might still want to look out for blips that could hurt after wearing the mask for a while if the user doesn't have a lot of hair.

The new solution of interspacing the lines better is not finished yet. I suspect it could also increase the segment count similarly.