Ultimaker/CuraEngine #1244: feature request: partialy retract and unretract durring travel - time saver

jmerc77, 2020-04-25:
using a min and max retract, initially retract by min, then to max during first half of travel.
then unretract back to min during last half of travel and finally, after travel finish unretract to "0.0".

this could be a huge time saver for prints with a lot of retracts.

the idea is that the first couple of mm of retract is used to prevent "blobbing", while the rest prevents oozing later in travel. since the later few mm's is only necessary a few hundred ms later,it can be combined with the travel to save print time.

rburema, 2020-04-28:
Have you tried this? (For example by editing gcode?) What where the results?

jmerc77, 2020-04-28:
i would like to but i don't know how to make the mod script yet (was thinking of using python). Testing solutions will be appreciated.

jmerc77, 2020-04-28:
getting ready to test! i figured out the test script...

Ghostkeeper, 2020-04-28:
I'd expect to see quite some oozing at the beginning of the travel and the end of the travel. Oozing is more of a function of the current pressure inside the nozzle chamber than anything else. While you're printing there is an overpressure in the nozzle chamber. Most of the retraction is to get rid of that overpressure. The rest is to cause enough of an underpressure to suck the material out of the nozzle opening.

jmerc77, 2020-04-28:
Ok, got more stringing. but I think it only the retract that matters, i'll try combining the un-retract but not the retract.

jmerc77, 2020-04-28:
another problem is when the travels are too short. the e speed is too fast on short travels. a way to fix that would be to only do this on longer travels.

jmerc77, 2020-04-28:
btw without mod I get a small amount of stringing. what i'm looking for is weather or not i get more than usual. also linear advance is enabled. maybe to get a better test I should set k=0.

jmerc77, 2020-04-28:
ok. seems about the same with some un-retract durring last half of travel. but I do need to adjust my profile some as even the control strings a bit.

jmerc77, 2020-04-28:
still very similar with k=0.

[image]
[image]

jmerc77, 2020-04-28:
saves 1 of 12 minutes in this case.

mahtDFR, 2020-05-15:
We probably aren't able to devote time to this right now, but we think there's some value to this. We'd like to see a pull request for it if you're able.

jmerc77, 2020-05-15:
if it involves me to write code in to the repo it may get messy. I perfer to leave it to the devs. I am not in a hurry to see this feature. so take the slow boat if needed. after all, the time saved seems to be small and varies on settings. also I realized that if there are multiple travel moves in one travel it may get tricky (like when trying to avoid printed areas).