Ultimaker/CuraEngine #1245: [4.6.0] slicer produces gap in wall

sjoerdtimmer, 2020-04-27:
**Application Version**
4.6 but I've seen it in the past too

**Platform**
All

**Display Driver**
Not important

**Steps to Reproduce**
Download Roller_insert_485.stl from https://www.thingiverse.com/thing:3209211/files
Slice wit 0.25 initial layer and 0.2 layer height

**Actual Results**
Layer 66 is disconnected from the previous layer except for a few small points where the wall touches the infill of the layer below. 

**Expected results**
Some extra plastic needs to be printed in layers 65 and below to make the model part above it connect to it. 

**Additional Information**
Skin expand settings don't fix this because there is no skin to expand. Maybe a concept like roof-wall should be introduced which counts as skin when generating the expanded skin area.
Alternate extra wall and print thin walls also don't affect this behaviour.

My current work-around is to set skin wall count to 5 but that shouldn't be necessary and for large models it would be very wasteful.

smartavionics, 2020-04-27:
A common solution to this problem is to include in the model a skin across the hole so that the smaller diameter upper part is not printed on air. After printing, the skin across the hole is removed.

Here is an example of doing that...

[image]

[image]

sjoerdtimmer, 2020-04-28:
It's certainly possible to work around this,
but wouldn't it be nice if cura sliced it correctly in the first place?

smartavionics, 2020-04-28:
How do you mean correctly? That model, like a lot of models is not really suitable for fdm printing. If you look at the mesh you can see it is made from two, overlapping, volumes. Using the cura mesh tools you can split the model into separate parts and then it does actually become easier to print...

[image]

[image]

sjoerdtimmer, 2020-04-28:
Sorry, my bad. Because it looked solid in the preview I had assumed that the model was correct.

closed 2020-04-28 by sjoerdtimmer