Ultimaker/CuraEngine #1246: How to understand some members of SliceLayerPart???

sscoder-3d, 2020-04-28:
Hello everyone, I am a beginner in cura, I don’t understand some members of SliceLayerPart and SkinPart, someone could help me?
1. SliceLayerPart.perimeter_gaps: The gaps between consecutive walls and between the inner wall and outer skin inset
What is outer skin inset?

2. SkinPart.insets: The skin can have perimeters so that the skin lines always start at a perimeter instead of in the middle of an infill cell
Why skin lines need to be inset?  I understand that skin is bottom / top ’s infill.

3. SkinPart.perimeter_gaps: The gaps between the extra skin walls and gaps between the outer skin wall and the inner part inset
What is extra skin walls? what is outer skin wall?  what inner part inset?

4. SkinPart.roofing_fill: The inner infill which has air directly above
why The inner infill has air directly above ?

Ghostkeeper, 2020-04-28:
You'll find that the terminology used in our code is not consistent by far. We tend to use the word "skin" to mean the top and bottom side of the print (indicated in yellow in layer view if you select the "line type" colour scheme). We tend to use the word "inset" to mean what Wikipedia calls a [parallel curve](https://en.wikipedia.org/wiki/Parallel_curve). The term "roofing" is used internally to mean "Top Surface Skin". This was renamed because "roofing" sounds too much like "roofies". See also our small [glossary](https://github.com/Ultimaker/CuraEngine/wiki/Glossary) page on the Wiki of this repository.

I'll answer your questions individually:

1. This refers to the extra walls around the skin if the Extra Skin Wall Count setting is `>= 1`, or just the outermost skin line if the skin is printed with the concentric pattern.
2. Eventually Cura needs to write the coordinates of the centreline of the Extra Skin Walls to the g-code. To get the centreline of a wall following the contour of a skin area, CuraEngine will perform an inset of half a line width. And if you have multiple Extra Skin Walls, it will perform another inset with a full line width for each. This gets you a contour that runs exactly through the middle. The vertices of this contour are the coordinates that get written to g-code to print it. The documentation here refers to the intended use of Extra Skin Wall Count: If you use the lines pattern for skin, these lines would then be attached to the Extra Skin Wall rather than end somewhere halfway inside the infill.
3. With "skin wall" it means the walls generated by the Extra Skin Wall Count setting again. If you have multiple of these, there can be gaps between them when the shape of your skin has sharp corners. The "outer skin wall" is then the outermost of these walls. With "inner part inset" it means the innermost (normal) wall. This neglects to mention that it also includes the gaps between the infill and the innermost wall.
4. Terminology is a bit messed up here. I wouldn't use the word "fill" here to prevent confusion. It refers to the part of the skin that is less than <the value of the Top Surface Skin Layers setting> layers underneath the top of the model. In other words, the volume of the Top Surface Skin Layers. With "air" they mean the part that is not inside the actual input model. This documentation assumes there is only one layer, but Top Surface Skin Layers can be multiple layers so it may not be "directly" underneath air.

sscoder-3d, 2020-04-30:
> You'll find that the terminology used in our code is not consistent by far. We tend to use the word "skin" to mean the top and bottom side of the print (indicated in yellow in layer view if you select the "line type" colour scheme). We tend to use the word "inset" to mean what Wikipedia calls a [parallel curve](https://en.wikipedia.org/wiki/Parallel_curve). The term "roofing" is used internally to mean "Top Surface Skin". This was renamed because "roofing" sounds too much like "roofies". See also our small [glossary](https://github.com/Ultimaker/CuraEngine/wiki/Glossary) page on the Wiki of this repository.
> 
> I'll answer your questions individually:
> 
> 1. This refers to the extra walls around the skin if the Extra Skin Wall Count setting is `>= 1`, or just the outermost skin line if the skin is printed with the concentric pattern.
> 2. Eventually Cura needs to write the coordinates of the centreline of the Extra Skin Walls to the g-code. To get the centreline of a wall following the contour of a skin area, CuraEngine will perform an inset of half a line width. And if you have multiple Extra Skin Walls, it will perform another inset with a full line width for each. This gets you a contour that runs exactly through the middle. The vertices of this contour are the coordinates that get written to g-code to print it. The documentation here refers to the intended use of Extra Skin Wall Count: If you use the lines pattern for skin, these lines would then be attached to the Extra Skin Wall rather than end somewhere halfway inside the infill.
> 3. With "skin wall" it means the walls generated by the Extra Skin Wall Count setting again. If you have multiple of these, there can be gaps between them when the shape of your skin has sharp corners. The "outer skin wall" is then the outermost of these walls. With "inner part inset" it means the innermost (normal) wall. This neglects to mention that it also includes the gaps between the infill and the innermost wall.
> 4. Terminology is a bit messed up here. I wouldn't use the word "fill" here to prevent confusion. It refers to the part of the skin that is less than  layers underneath the top of the model. In other words, the volume of the Top Surface Skin Layers. With "air" they mean the part that is not inside the actual input model. This documentation assumes there is only one layer, but Top Surface Skin Layers can be multiple layers so it may not be "directly" underneath air.

thks！

closed 2020-04-30 by sscoder-3d