Ultimaker/CuraEngine #1256: Question: Using Cura for 5axes 3dprinter

l1va, 2020-05-24:
Hi, 
Is it difficult to change sources or by plugins or with something else to add support  5axes 3d printing. I mean rotating and tilting platform (bed). Is it possible to slice not by Z? 
If yes, point please where it is better to start, what code should be updated firstly. 
By fast review seems it is not so easy - all algorithms use such base that slicing is always by Z.

rburema, 2020-05-24:
This is going to be quite difficult, perhaps impossible without rewriting an enormous amount of code.

The only proposed way of doing it that I have heard that is somewhat feasible and comes even close to what you want -- is to 'warp' the 3D-model so it can be sliced, then perform the opposite transformation on the sliced layers (back to 'real world space').

l1va, 2020-05-24:
Thanks for the idea! 
But it seems it is not so easy, works not always and I am afraid problems with visualization...
Understand. 

I will wait some days, might be somebody has other ideas, and then will close the issue.

nallath, 2020-05-25:
I think you will need a bit more than "just" plugins to do this. Slicing for a 5 axis machine is fundamentally different from what is done now. In order to support it, you would probably have to re-implement the entire slicing engine.

Ghostkeeper, 2020-05-26:
First, you would expect it to not work with planar layers but perhaps spherical or cylindrical layers. Or just variable-thickness layers in general.
You will also run into major issues with support generation and bridging, which would work completely differently then.
The whole idea of what is a wall and what is a skin and what should the thicknesses be would become muddy.

This is out of scope for us, definitely.

closed 2020-05-26 by Ghostkeeper