Ultimaker/CuraEngine #1258: Want to hide the cooling fan stop command (M107) in the output gcode.

narutorne, 2020-05-28:
Thank you for seeing.
And always thanks to the cura community.

I am a beginner, 
and my English isn't enough good,  so feel free to ask me if there is anything unclear.

I am trying to slice with CuraEngine on the command line. (Slicing is fine.)
I would like to know how to hide the cooling fan stop command (M107) in the output gcode.

I'm thinking of operating a not normal 3D printer

In the output gcode,
I don't want commands about temperature and cooling fan.
Trying to make it non-output.

Regarding the temperature item,
"machine_nozzle_temp_enabled" to false,
About turning on the cooling fan,
I was able to hide it by setting "cool_fan_enabled" to false,

But I didn't know how to non-output M107, which is a cooling fan stop command.

I would be grateful if you know how to solve this.
Thank you for reading this far.

Ghostkeeper, 2020-06-02:
No, CuraEngine will currently always output `M106` or `M107`, or `M126`/`M127` for the Makerbot g-code flavour. There's no way to prevent that. Cura's front-end can remove them with a post-processing script though, so you could try that, or writing your own post-processing script. Basically you just need to run a search and replace on the g-code string:

```
with open("my_print.gcode") as f:
    gcode_str = f.read()
gcode_str = gcode_str.replace("M107\n", ";M107\n")
with open("modified.gcode", "w") as f:
    f.write(gcode_str)
```

I don't think we'll implement a feature to prevent writing fan speed commands in CuraEngine though. I'd expect that printers that don't have any fans would just ignore the M107 in their firmware.

closed 2020-06-02 by Ghostkeeper

narutorne, 2020-06-06:
Thank you for your reply.
I understand there is no function that M107 does not output, (but it is not necessary)
>> I'd expect that printers that don't have any fans would just ignore the M107 in their firmware.

And I also understand that it is possible to deal with cura front end or original post-processing script.

Thank you for your reply to such a trivial issue.