Ultimaker/CuraEngine #1259: [ERROR] setting with no value given: 'machine_extruder_cooling_fan_number'

CutAs, 2020-05-29:
Hi, 

I'm trying to use curaEngine on a Mac, the one that comes packaged inside Cura UI contents.

./CuraEngine slice -v -j /[default definitions directory]/ultimaker_s3_MOD.def.json -o "[random directory]/slice_test/output_test.gcode" -e0 -l "[random directory]/stl_example_binary.stl"

Seems to load all settings fine, including machine_extruder_cooling_fan_number=0, then:

Starting export...
[ERROR] Trying to retrieve setting with no value given: 'machine_extruder_cooling_fan_number'

What am I doing wrong? Thanks.

Ghostkeeper, 2020-06-02:
We've noticed this as well in debugging. It doesn't properly resolve the inheritance in the printer definitions. It depends on the current working directory.

CutAs, 2020-06-02:
Yep, I spent quite a few days on this and indeed it 1) does locate and load all the inherited def.json files in the path correctly, 2) does not properly override the values. 

Not sure if this helps as I'm not much of a debugger, but what I noticed is that when I appended machine_extruder_cooling_fan_number=0 at the bottom of all the overrides in ultimaker_s3.def.json, it still didn't read the value. When I moved it higher up the list of overrides (randomly, unfortunatelly), it read the setting. Could it be related to the alphabetical reading order of the settings? Noob guess.

CutAs, 2020-06-02:
I also noticed that ultimaker_s3_extruder_(left/right).def.json contain:
"machine_nozzle_head_distance": { "default_value": 2.7 }, a value which is not present in fdmextruder.def.json

Dystopi, 2020-09-04:
I was able to work past this issue by specifying the extruder def.json file as an additional argument to the cli command

```bash
curaengine slice -v -j slicer/resources/cura/definitions/fdmprinter.def.json -j slicer/resources/cura/definitions/fdmextruder.def.json -o "slicer/test.gcode" -e1 -s infill_line_distance=0 -e0 -l "slicer/resources/stl_files/Manticore_-_No_Base.stl"
```

Lafinger, 2022-04-16:
I also faced this problem.How can I solve this problem?

Unknowncaberu, 2023-01-18:
You have to put : -s "machine_extruder_cooling_fan_number"=value 
As example : CuraEngine slice -v -p -j jsonfile.def.json -s "machine_extruder_cooling_fan_number"=1 -o "file.gcode" -l "file.stl" 
If it doesn't work, use just : -s machine_extruder_cooling_fan_number=1