Ultimaker/CuraEngine #1260: [Warning] No default value set for:

CutAs, 2020-05-30:
In my command line, I'm loading -j Ultimaker_S3.def.json and -j ultimaker_S3_left_extruder.def.json

Technically they are inheriting settings from generic ultimaker, fdmprinter, and fdmextruder

For some reason, if I load only fdmprinter and fdmextruder, I receive no warnings. If however, I try to override settings with ultimaker_s3 definitions, around 100 values are not read properly (warning: no default value). 

What am I doing wrong? Is there a way to read the def.json correctly? 

Thanks.

nallath, 2020-06-02:
The Ultimaker_s3 inherits some settings from other profiles. Did you check if the engine could also find those (eg; The Ultimaker_s3 inherits some stuff from "ultimaker.def.json", which in turn inherits from "fdmprinter.def.json")

I personally just use the CLI interface with fdmprinter.def.json, so I'm not sure if the inheritance bit works entirely how it should.

CutAs, 2020-06-02:
Hi, yes I did. In fact, slicer throws an error if one of the inherited files is missing from the path. So it gets the files but doesn't properly read the values.

CutAs, 2020-06-02:
My only workaround so far was to manually hard code the settings from ultimaker_s3.def.json (and extruder) to fdmprinter.def.json (and extruder) and then slice with those.

Ghostkeeper, 2020-06-03:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/1259, though you're now seeing a different setting that has a different outcome.

closed 2020-06-03 by Ghostkeeper