Ultimaker/CuraEngine #1261: 'Something is wrong with this file' Ultimaker_S3 bad gcode header.

CutAs, 2020-06-02:
Ok promise my last question, then I will stop spamming. 

I have a .gcode sliced via curaEngine but unfortunately it does not contain all the values needed for a correct gcode header. The missing ones are:

— Material ID
— Print Time
— Filament Amount

For the material ID, I pass it as a slicer setting and it gets read but for some reason does not go into the header? 

For the later two, I came up with this 'dirty' plan of saving terminal output to a .txt, grabbing the last three lines that get printed after 'END OF GCODE' via bash or python, then appending them (including material ID) to the .gcode file. 

Is that the best one can do these days or is there a more elegant solution cos it's 2k20?

Ghostkeeper, 2020-06-03:
For the material GUID, please check if the extruder count is set correctly. Since the GUIDs are obtained in the code with a simple array access, it's checking first if the length of the list of GUIDs is equal to the extruder count.

Printing time and used filament are expected behaviour when working with the command line debugging interface. This is because to prevent holding all of the g-code in memory before outputting everything, it's outputting g-code as it gets generated. But since the header needs to be output first, it doesn't know yet at that point how much material is used or how long the print takes. It's somewhat inherent to the way this works. I don't think we'll want to change that just for debugging, since that would affect the normal GUI workflow as well.

CutAs, 2020-06-04:
Hi, thanks for the tips. Yes, the extruder count is correct. But nevermind, since I had to do the workaround python script I just parse the material from there as well. 

Re: rest. Makes total sense. I'm actually using curaengine for an art installation in which GUI is not really an option. I know my case might be one in a million but a 3d printer, besides being functional, is more than often a creative tool. Therefore, leaving modular interfacing options other than user-friendly front-ends available is a very welcome decision on your side. Hope you keep it that way in the future!

Ghostkeeper, 2020-06-05:
We're not planning on removing the command line interface of CuraEngine, don't worry! But it doesn't get a whole lot of attention since for our product it's only a debugging tool. We won't make any changes that hurt the performance of the GUI in order to fix something with the CLI.

CirnoGiovanna, 2020-12-28:
Hi, @CutAs,  I have another solution but I'm not sure is it elegant. A few days ago, I met the same problem in the same type of machine (Ultimaker S3). By combining this issue and knowledge in [https://community.ultimaker.com/topic/15555-inside-the-ultimaker-3-day-1-gcode/](url), I find that the gcode header needs these keywords to execute in Ultimaker S3, but their values need not be exactly right. Thus, I modified gcodeExport.cpp (line 190~220) to write fixed MATERIAL.GUID, PRINT.TIME, and VOLUME_USED into gcode header. Then, the rebuild CuraEngine can generate an Ultimaker S3-execuable gcode once.

Ghostkeeper, 2020-12-28:
Ah, stupid of that firmware. I was under the impression that the Ultimaker S3 would still print g-code generated by other slicers and would have accept g-code even if the header is missing or incomplete. That's what the Ultimaker 3 does anyway... :(

acpjcj, 2021-04-28:
Hi! I have the same problem with ultimaker 2+ connect. 
I tried print logs to follow the script because I thought the problem was in the command line or in the profile of impression. I don’t understand why the app in desktop makes gcode correct and the command line makes wrong gcode.

Ghostkeeper, 2021-05-13:
> I don’t understand why the app in desktop makes gcode correct and the command line makes wrong gcode.

The reason for the time and material estimates is that those statistics about the g-code output are calculated as the g-code output is being generated. When slicing via the front-end, the g-code is output in an I/O stream through Arcus, and the header is sent separately afterwards. When slicing via the command line, the g-code is output to stdcout. This means that when slicing via the command line, the g-code header is not yet known by the time it needs to be put in the output stream. It is then filled with dummy estimates, since it's only meant for debugging.

It can be solved with a refactor, outputting the g-code first to a string buffer, then outputting the header to stdcout and then the string buffer to stdcout. But we're not about to do that refactor for just the CLI debugging. It's not important for the estimates to be 100% correct there.