Ultimaker/CuraEngine #1270: Subvolume division of 3D model

abtionlove, 2020-06-08:
Does the latest version of curaengine have the function of decomposing 3D model, dividing sub volume and slicing?

nallath, 2020-06-08:
Could you explain what you mean a bit more extensive? I can think of a number of different interpretations.

abtionlove, 2020-06-15:
> 您能解释一下您的意思吗？我可以想到许多不同的解释。

Whether curaengine divides the complex 3D model into subvolumes, and then processes the subvolumes separately.

nallath, 2020-06-15:
It can't do this automatically, but you can do this by using "modifier meshes". An example of that can be seen on https://www.youtube.com/watch?v=PUwf79wySUY

closed 2020-06-15 by nallath

abtionlove, 2020-06-15:
> It can't do this automatically, but you can do this by using "modifier meshes". An example of that can be seen on https://www.youtube.com/watch?v=PUwf79wySUY

Hello, first of all, thank you. Can you elaborate? I can't understand. I can't open the website you gave me.

nallath, 2020-06-15:
Can you access this? https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura

abtionlove, 2020-06-15:
> Can you access this? https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura

Yes, thank you very much