Ultimaker/CuraEngine #1276: Retrieve filament and print time from CuraEngine in a Node.js App

emrem6, 2020-06-16:
CuraEngine is successfully running and gcode is generating. But is there a way to retrieve in node.js the following data?

- Print time (s): 

- Print time (hr|min|s): 

- Filament (mm^3): 

I want to code an App which is calculating the costs of a print order. Therefore I need to have access to these information. But the gcode file doesn't contain these information and I don't know how to pick it from the command-line from my App.
Any ideas on that?

Ghostkeeper, 2020-06-17:
The only way to get that from CuraEngine is to call `./CuraEngine connect` rather than `./CuraEngine slice`. It sends the result over Protobuf, so you'd need to use a language that supports Protobuf. JavaScript is not supported by Protobuf, unfortunately.

Alternatively, I once ported CuraEngine's time estimation algorithm to Python. It can load g-code in and estimates the printing time based on that. It's slightly different since it doesn't do temperature regulation so it can't model waiting for the nozzle or build plate to heat up, but otherwise it's the same as the state of CuraEngine's time estimation 2 years ago or so (which is 99.9% the same as now): https://github.com/Ultimaker/Cura/blob/master/scripts/check_gcode_buffer.py
Only that script calculates the peak segments per second that the CPU has to process. You don't need that result, so you'd need to modify it still.

It doesn't do filament usage calculations. Filament is much easier though: Just run through the g-code and sum up all increases in E value (but do take care of absolute vs. relative extrusion mode).

charkops, 2020-07-28:
You can pipe the result of sterr (most info are on the error stream instead of the out stream, not sure why) to a string and manually try to find the "Print Time(s):" && "Filament (mm^3):" lines, parse them and get the results. 
Do take care with failed slices though

example code used in one of my projects for the exact same thing: 

``` c++
let result = spawnSync('CuraEngine', [
      'slice',
      '-v',
      '-j',
      pathToCuraProfile,
      '-l',
      pathToStl,
      '-o',
      '/dev/null',
    ], {
      stdio: ['pipe', 'pipe', 'pipe']
    }).stderr.toString();
```

You can then go through the result line by line and search for the info you want.

Ghostkeeper, 2020-07-30:
Ah yeah, if you use `-v` it'll output the print time and material usage in stderr. Good point.

closed 2020-07-30 by Ghostkeeper