Ultimaker/CuraEngine #1277: Feature Request: Option to insert M73 (Set Print Progress) in generated GCODE

marciot, 2020-06-16:
Currently Cura does not output M73 commands to the generated GCODE, which is helpful for updating the progress bar in a meaningful way.

I can see two ways to implement this feature. Either add a flag which enables the generation of M73, or possibly more generally, add a GCODE template that is emitted at each layer with a special placeholder variable for the progress value. That way, someone could set that script to output a M73 if they desired.

Ghostkeeper, 2020-06-18:
Feels like there's a dozen and one standards for this. M532, M73, M117...

But yeah, I think this is something that could be added. I'll propose it to the team.

mahtDFR, 2020-06-18:
This was discussed this morning in our standup. It's a valid feature request but as something that will be used by third-party printers and not Ultimaker printers, it's hard to prioritise right now, so we've had to defer it. But we'd still like to see it and are happy to review pull requests that implement it.

marciot, 2020-06-18:
Okay. Thanks for considering it.

KimmoHop, 2020-09-08:
See also [issue in Cura](https://github.com/Ultimaker/Cura/issues/8331).
Post-process script to display M73 P progress and M117 remaining time already exists. Progress is updated every percent, remaining time every layer. It would be possible to use M73 R and more frequent update for remaining time.

Time placeholders (maybe every minute in addition to layer change) from time estimator or similar source would be a bit more accurate and a lot easier to convert in post-processor to printer format. Maybe I'll have to look, though last time I built Cura Engine it didn't run ;)