Ultimaker/CuraEngine #1279: [ERROR] setting with no value 'machine_always_write_active_tool'

CutAs, 2020-06-25:
Just built a fresh curaEngine on Ubuntu and no matter what I do after 'starting export...' I immediately get 
[ERROR] Trying to retrieve setting with no value given: "machine_always_write_active_tool" 

I have curaEngine built a couple of weeks ago on a similar machine and it all runs fine. 

Did I miss something when building? 

Thanks.

ninovanhooff, 2020-06-25:
Note: this setting was introduced on may 7th. PR https://github.com/Ultimaker/CuraEngine/pull/1197

Might help with findind the problem

CutAs, 2020-06-26:
> Note: this setting was introduced on may 7th. PR #1197
> 
> Might help with finding the problem

Ah, I see, thanks! I took my preset definition files from the older build so my fdmprinter.def.json is not the latest version and probably misses this and potentially some other settings. Will give it a shot.

closed 2020-06-26 by CutAs