Ultimaker/CuraEngine #1292: [ERROR] Couldn't find extruder.def.json file for extruder 0

callmebg, 2020-07-14:
Help me，please！
When i build and run curaengine.exe in terminal, i get some bugs. I sure the json and stl exist.
What should i do? 
or Who can give me a project that can run successfully in vs 2017?
[image]

rburema, 2020-07-15:
You should be able to get those definition (.json) files from the main Cura repository for different printers. Replicate the folder structure found there (`./definitions`, `./extruders`), and you should be good. This is because the fdmprinter file refers to fdmextruder (file).

callmebg, 2020-07-15:
> 
> 
> You should be able to get those definition (.json) files from the main Cura repository for different printers. Replicate the folder structure found there (`./definitions`, `./extruders`), and you should be good. This is because the fdmprinter file refers to fdmextruder (file).

cool！it works！thank you

closed 2020-07-15 by callmebg