Ultimaker/CuraEngine #1293: Certain slicing settings not being applied when using curaengine from command line

simar1998, 2020-07-14:
Greetings, I am having an odd issue with cura egine when executing a slice command from the command line. Certain things like the layer height and bed adhesion type are not being applied to the gcode file. For example.

If I execute this command 

`slice -v -p -j /home/simar/def/creality_cr20.def.json -l /home/simar/cube.stl -s layer_height_0=0.3 -s layer_height=0.2 -s line_width=0.4 -s infill_sparse_density=20 -s infill_pattern=grid -s wall_thickness=0.8 -s top_bottom_thickness=0.8 -s ironing_enabled=false -s retraction_speed=25.0 -s material_print_temperature=210.0 -s retraction_amount=6.5 -s material_initial_print_temperature=215.0 -s material_bed_temperature=60.0 -s material_flow=100.0 -s cool_fan_speed_0=100.0 -s cool_fan_enabled=true -s adhesion_type=brim -s speed_print=60.0 -s speed_travel=120.0 -s speed_support=60.0 -s speed_wall=30.0 -s speed_infill=60.0 -s speed_topbottom=30.0 -s support_type=everywhere -s support_enable=false -s support_pattern=zigzag -s support_angle=50 -s support_infill_sparse_thickness=0.1  -s center_object=true    -o /home/simar/test.gcode`

Layer height always stays at 0.1 and bed adhesion is always brim. Is it possible that the properties from the def file is somehow overwriting whatever i am specifying from the command line ?

rburema, 2020-07-15:
`-s adhesion_type=brim` is literally in your command ... I'd expect brim to come out?

Anyway, I think the (`-l` and) input file needs to be at the end of the command, IIRC, or at least after all the settings.

simar1998, 2020-07-15:
@rburema Sorry for that, however even if I put something else for example "none" in the adhesion_type, brim shoes up

Ghostkeeper, 2020-07-16:
Yes, put the `-l` more at the end.

The problem is that the syntax is not very clear here. Every setting changed after the `-l` will be applied as a per-object setting on the model that you're loading. Since the adhesion type and the layer height can't be set per-object, those settings are ignored.

closed 2020-07-16 by Ghostkeeper

simar1998, 2020-07-16:
> Yes, put the `-l` more at the end.
> 
> The problem is that the syntax is not very clear here. Every setting changed after the `-l` will be applied as a per-object setting on the model that you're loading. Since the adhesion type and the layer height can't be set per-object, those settings are ignored (as well as those temperatures and bed temperature, retraction distance, etc.)
> 
> `slice -v -p -j /home/simar/def/creality_cr20.def.json -s layer_height_0=0.3 -s layer_height=0.2 -s line_width=0.4 -s infill_sparse_density=20 -s infill_pattern=grid -s wall_thickness=0.8 -s top_bottom_thickness=0.8 -s ironing_enabled=false -s retraction_speed=25.0 -s material_print_temperature=210.0 -s retraction_amount=6.5 -s material_initial_print_temperature=215.0 -s material_bed_temperature=60.0 -s material_flow=100.0 -s cool_fan_speed_0=100.0 -s cool_fan_enabled=true -s adhesion_type=brim -s speed_print=60.0 -s speed_travel=120.0 -s speed_support=60.0 -s speed_wall=30.0 -s speed_infill=60.0 -s speed_topbottom=30.0 -s support_type=everywhere -s support_enable=false -s support_pattern=zigzag -s support_angle=50 -s support_infill_sparse_thickness=0.1 -s center_object=true -l /home/simar/cube.stl -o /home/simar/test.gcode`

That worked perfectly, thank you!