Ultimaker/CuraEngine #1308: [4.6.2] Retract before or during Coast and wipe in Reverse

AxlePivot, 2020-08-18:
Thank you for using Cura!
-->

**Application Version**
 4.6.2

**Platform**
 win 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2020-08-19:
Retracting during a printing move is generally a bad idea. The acceleration limit kicks in really quickly then, and causes the X/Y movement speed of the nozzle to slow down. This results in a blob, in your case on the outer wall. It really does more harm than good.

Making the Outer Wall Wipe or the Infill Wipe reverse over its own path would defeat their primary purpose. The primary purpose of the Outer Wall Wipe is to hide the seam, so reversing would introduce a deceleration to stand-still and then acceleration from stand-still again, producing a big seam. The primary purpose of the Infill Wipe is to make the infill adhere better to the walls, so making it reverse on itself causes the bead to be dragged back and then not stick to the wall any more, rather than stick better. Their purpose is not to reduce ooze, but I suppose it does that too to an extent.

I think slowing down your retraction would help a lot to reduce stringing. 0.1s for 8mm is extremely fast for a retraction, especially with a Bowden tube. Depending on your acceleration and jerk settings, that would have to be a maximum speed of upwards to 160mm/s then? It's going to give a lot of slip. And in 0.1s it's not going to have enough time to stop the material from flowing either. The retraction is happening in a safe place away from the outer wall, so feel free to slow that down a lot. Any lost material can be recouperated with Retraction Extra Prime Amount if you feel like you're getting underextrusion after a retraction.

AxlePivot, 2020-08-28:
Hi, Thanks for the response and sorry for the delay but I have been looking at your suggestions and doing some more trials. Note i did realise that 0.1s/8mm was fast and the point i was trying to make but badly was that the nozzle will be static for at least 0.1s but probably longer. I tested retraction with bowden and to get oozing to stop a retration of 10 to 13 mm can help, at this point i believe the slightly fattened heated end of the filalment is acting like a syringe in the PTFE creating a small vacuum preventing flow from the nozzle. I also tried printing upside down and this has no noticable impact so ooze isnt due to gravity either! Ooze must therfore be determined by the material and heating effects such as thermal expansion and moisture content causiing expansion ion the nozzle itself once the unmelted filament has been withdrawn.

The  challenge I have is need to print large parts (4 day print) cosisting an array of small hollow sections (5mm square with only 2 wall counts) so often there is no infill hence the nozzle pauses on the shell while it retracts. This is why i am looking for a feature to begin retract at the end of the perimiter line. The alternative is to move into clear internal space to allow the nozzle to ooze in safe space even though I think this option would risk excesive build up of loose material on the nozzle. The feature does exist in PrusaSlicer with 'Wipe while Retracting' . I therefore tried this in PrusaSlicer and managed to reduce stringing compared to not having this on or in Cura. I would rather use Cura though if the feature can be added as it is a better interface overall from my testing!

The second issue I have with my part printing is there are numerous external moves and again the ooze is a problem during travel as it catches on the shell and causes the little 'tree branch' features. To overcome this I use Z hop so the blob is landed onto the new start point rather than catching randomly on the edge. The problem with Z Hop it increases stringing as the filament is drawn out as it lifts etc. I have overcome this by manually (excel macro) modifying z hop to occur during the travel to the next feature, using G1 for interpolation. The nozzle therfore leaves the last print area nearly horizontally and land vertically. I conducted a set of tials and attached is the results that show how effective this is. It is also very smooth motion as following the 'Wipe while Retract' there is no sudden Z direction change and the move can happen more quickly but still provide Z hop landing.

Therefore the killer features I would be hoping for is 'retract during wipe' and 'z hop during travel'

I have attached a copy of my notes and photos of the results as I think this could be a great way to help reduce stringing and blobs.
Happy to add more comment detail if this doesnt make scense or if you have another idea how to fix the problem. Its a great product and I would love to help.
[image]
[image]
[image]