Ultimaker/CuraEngine #1315: how to use def.json file in cura engine

Thrada, 2020-08-30:
Application Version
4.5

Platform
Ubuntu 24.04

I made def.json file which inherit fdmprinter.def.json

I wrote override values in json file like this:

"overrides": {
            "material_diameter": { "default_value": 1.75 },
            "machine_nozzle_size": { "default_value": 0.4 },

            "layer_height": { "value": 0.2 } ,
            "layer_height_0": { "value": 0.3 } ,
            "line_width": { "value": 0.4 } ,

            "infill_line_distance": { "value": 2.66 } ,
            "infill_overlap": { "value": 15 } ,
            "infill_before_walls": { "value": false } ,

            "default_material_print_temperature": { "value": 180 } ,
            
            "wall_line_count": { "value": 2 } ,
            "top_layers": { "value": 4 } ,
            "bottom_layers": { "value": 4 } ,

            "speed_print": { "value": 50.0 } ,
            "speed_infill": { "value": 50.0 } ,
            "speed_wall": { "value": 50.0 } ,
            "speed_wall_0": { "value": 50.0 } ,
            "speed_wall_x": { "value": 50.0 } ,
            "speed_topbottom": { "value": 50.0 } ,
            "speed_travel": { "value": 50.0 } ,
            "speed_layer_0": { "value": 10.0 } ,
            "speed_slowdown_layers": { "value": 4 } 
}

I excuted command like this:
slice -v -j def/myprinter.def.json -o ~/result.gcode -l test.stl

after in result.gcode it shows M104 S215 in front gcode(I wrote default_material_print_temperature value to 180)

is there something wrong in my command?

sorry for my bad english

Ghostkeeper, 2020-09-01:
As of currently, it needs to find the printer it inherits from in the working directory. So you'd need to execute that command from the directory that both `myprinter.def.json` and `fdmprinter.def.json` are located in. The debugging routine is a little bit dumb with how it figures out paths to the `inherits` metadata field value.

Ghostkeeper, 2020-09-01:
Try adjusting `material_print_temperature_layer_0` to adjust the first layer.

And try adjusting `material_print_temperature` to change the actual printing temperature. That `default_` version is only used in the front-end to allow material profiles to define a printing temperature as well as a quality profile. Technical details... But it's not actually used for slicing.

closed 2020-09-01 by Ghostkeeper