Ultimaker/CuraEngine #1329: Something strange hole at wall

joonghochoe, 2020-09-23:
**Application Version**

master version

**Platform**

windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

1. load housing.3mf
2. change "Maximum Resolution" to 0.2
3. change "Maximum Travel Resolution" to 0.6
4. change "Maximum Deviation" to 0.1
5. slice
 
**Actual Results**

there is some strange hole at wall

**Expected results**

no hole

**Additional Information**

[image]

3d model file is

[UMS5_housing.zip](https://github.com/Ultimaker/CuraEngine/files/5266259/UMS5_housing.zip)

there is no strange hole in 4.7.1 version. It causes abnormalities from the cura engine merged version on September 17, "Merge pull request #1321 from Ultimaker/CURA-7709_simply_leaves_tiny_segments"

Ghostkeeper, 2020-09-23:
I'd be surprised if #1321 itself would cause this since that algorithm only adjusts the vertices of a contour. It's not able to create new contours.
Could it be changes we made to stitching a few days earlier? https://github.com/Ultimaker/CuraEngine/commit/3b310a5ace1ac40ed77a21e6feb724295f8fcabc We already have an incoming fix for that here: https://github.com/Ultimaker/CuraEngine/pull/1325

joonghochoe, 2020-09-24:
A similar problem appears in the new master version(Sep.24).
Resolution Setting Changed of master version :
[image]

Resolution Setting Default of master version:
[image]

Resolution Setting Changed of 4.7.1 version:
[image]

Used 3D model 3mf file:
[UMS3_Compressor_Wheel_hole.zip](https://github.com/Ultimaker/CuraEngine/files/5279695/UMS3_Compressor_Wheel_hole.zip)

Ghostkeeper, 2020-09-25:
That project file reproduces the problem for me as well. Since the fix for stitching was already merged in #1325 I don't think that was the issue. We'll investigate.

konskarm, 2020-09-30:
Can you try with this fix https://github.com/Ultimaker/CuraEngine/pull/1336? Trying it locally, I see that the weird artifacts are no longer generated after changing the Maximum Resolution and Deviation.

joonghochoe, 2020-10-05:
The problem doesn't seem to occur in the new master version. Thanks.

closed 2020-10-05 by konskarm