Ultimaker/CuraEngine #1332: Feature Request: Extend prime tower functionality to be used as single material sacrificial cooling buffer.

Mike111177, 2020-09-25:
### Problem: 
When working with very fine layer heights (0.04mm), radiant heat from the nozzle frequently will melt layers below the one currently being printed. This problem is exacerbated by tall thin objects on your model that trip the minimum layer time, causing the nozzle to hover in the same spot for even longer, such as in the picture below.
[image]
The first thing I tried was to increase the minimum speed, this had limited effect as I feel the problem transitioned from having to much time near the nozzle, to having not enough time for each layer to cool. Next thing I tried was to use the lift head feature, to enforce the minimum layer time. This made excellent improvement on the sagging, but caused an unacceptable amount stringing ruining thin parts of the model. So I need some kind of tower to wipe the nozzle on while the layer is cooling down.

### My current workaround:
So I have gotten the best results by creating a sacrificial cooling buffer with the following procedure:
1. Create a cylinder with the height of the model, 0 infill, 2 walls thick, double wall print speed (quality doesn't matter for this cylinder), 0 top/bottom thickness.
2. On the cylinder In problematically slow layers place a box of mesh type: modify settings for overlaps, set print speed to the minimum print speed.

Here is what that ends up looking like in Cura:
[image]

The downsides to this is that its pretty difficult to finely adjust each layer speed due to to you having to make a new cube (in my case with a plugin), manually fix the infill and top/bottom thickness and speed for each cube, drag it up and down, scale it, slice a bunch of times (this also seems to dramatically increase the time it takes to slice). Its a pain and you can't really get it perfect for each layer anyway. Also, if you don't set the entire cylinder to min speed, sometimes you will get visible layer lines where the speed changes suddenly layer to layer. (For example the top and bottom of the torso of this mini). Kind of ruins the point of doing 0.04mm layers lol.

### Proposed Solution
After I did all that, I had an epiphany. Cura already has a feature to auto generate hollow towers meant for wiping the nozzles that have dynamic properties based on each layer. Prime Towers! They even can have an accelerated speed by default! So correct me if I am about to make a wrong assumption, I am still studying this repo, but I think it could be an incredibly easy change to allow prime towers to fill this role. I think that all that needs to be done is the following:

1. Allow prime towers to be generated on single extruder machines. They would simply be hollow since you don't need to purge right? (Maybe move them into their own category outside of dual extrusion, however I would not know if that would be a breaking change)
2. Add the following settings in the cooling section:
    - Reduce Prime Tower Speed First: visible when prime towers are enabled, default enabled.
       Modifies the behavior of `void ExtruderPlan::forceMinimalLayerTime(...)` to attempt to reduce the speed of the prime tower(s) to meet the minimum layer time before it reduces the speed of the rest of the model. If it can meet the min layer time without touching the speed of the model itself just using the prime tower paths, don't bother slowing down the paths in the model.
    - Minimum Prime tower speed: visible if Reduce Prime Tower Speed First enabled, default same as Minimum speed.
       Allow the minimum speed of the prime tower to be lower than the minimum speed of the model. Since we don't care about the quality of the prime tower, having a lower minimum speed for it would give a better chance to not have to touch the speeds of the model itself.

Those two things combined is basically the easiest way I can think to implement this. I am wondering what your thoughts are, if there is something I am missing or something that makes this not make sense. 

### Additional thought: 
Another cool thing maybe to do along this for quality of life would be to have the lift head feature invisible when  Reduce Prime Tower Speed First is enabled, but show another option, like "Park head" which will instead of Z hopping will just park the head next to the prime tower after each layer and then do an extra quick wipe before going back to the model. This would allow the slicer to guarantee that the model speeds wont change. but you can do the same thing by just slightly increasing your prime tower size or reducing your prime tower min speed so its not critical to the feature.

Ghostkeeper, 2020-09-29:
Nice post! Clear reasoning and you did your research.

There is a point at which the extra heat provided by reducing speed just causes more blobbing than what would happen if you were to just stack molten layers on top of each other without speed reduction. This is something that our print process engineers still want to address at some point. It may be better to not reduce the speed but actually _increase_ it with shorter layers, to reduce the amount of heat going into the print.

Did you try using a draft shield? It's a bit similar to a prime tower in that it prints something extra on every layer.

The prime tower is geared towards making sure the material is primed at every extruder switch, so it has a number of things that make it inappropriate for use as a delay to allow the print to cool down. For instance it's only printed up to the height of the last extruder switch. And it only primes the extruders that are going to be used on the next layer (aside from the structural extruder used for the prime tower outer shell). We could strip out all of those behaviours if you're using it for minimum layer time, but it's a bit clunky and shouldn't really be called a prime tower any more at that point.
The draft shield is geared towards protecting your print from drafts in the room. But it has fewer limitations, basically applies to every print. The only thing is that you can't tune the volume of material used in the shield so there's no way to tune the time it spends on it. And of course it also doesn't do all of the speed adjustments you propose so it could make your print take a little bit longer than absolutely necessary.

Mike111177, 2020-09-29:
> There is a point at which the extra heat provided by reducing speed just causes more blobbing than what would happen if you were to just stack molten layers on top of each other without speed [...]

That is an interesting point. So it sounds like there is a couple of adjacent goals here. 

1. The original concept that with shorter (timewise) layers you want to reduce to speed of extrusion in order to allow more time for that layer to cool before going to the next one. This makes sure that the layer has sufficient time to cool before experiencing the load/heat of the next layer on top of it. I believe this still makes sense. The direct downside to this is you may get increased blobbing during the slow down. Also it is directly antithetical to the next goal.

2. The new problem where on single smaller cross sections (there may be multiple on a single layer) since the heat does not disperse as quickly to the model below, meaning radiant heat from the nozzle is problematic. One solution could be to increase the print speed on these smaller cross sections to get the nozzle away from it faster. Also note that the "Lift Head" feature should also solve this implicitly, however that causes insane amounts of stringing.

3. There is also an argument that you would want to maintain a constant speed setting through each layer of the model to get consistent wall quality, and reduce noticeable layer lines. This obviously is not possible when you increase or decrease the speed.

For the rest of this I am going to reference these goals as **G1**, **G2**, and **G3** respectively. Also note for both of these goals, the common idea is to reduce the average amount of heat transfer/time for any given point of the model.  

> Did you try using a draft shield? It's a bit similar to a prime tower in that it prints something extra on every layer.

I haven't test printed it, but I sliced it and here are my thoughts.

[image]

I definitely like the fact the draft shield is already printed slower than the model, meaning for a large portion of it, the model speed is unaffected. However when it gets to the top of this model, it is still not enough to prevent the slow down (and probably sag) at the top of the flame in her palm. This violates **G2**. Also for less warpy materials, I'd rather not have the draft shield so I can get better cooling. This is all in addition to the limitations you stated.

I then emulated the same effect this with a single walled cylinder with a lower speed of 10mm/s and I like the results, with a lower print time and material cost.
[image]
The obvious problem here is that this only works because it is actually breaking the functionality of `void ExtruderPlan::forceMinimalLayerTime(...)` since this means the the smallest speed ends up equaling `minimalSpeed`, forcing `factor` to equal 1 meaning there is no speed change. This also means that I am no longer meeting my minimum layer time and I am potentially violating **G1**. I could mitigate that by increasing the volume of the cylinder, but there is no way currently to find out how much volume would be required. It also means the print time will take longer when the cylinder is overkill for the layer.

> The prime tower is geared towards making sure the material is primed at every extruder switch, so it has a number of things that make it inappropriate for use as a delay to allow the print to cool down. For instance it's only printed up to the height of the last extruder switch. 

I think, if a new cooling tower feature would be introduced it's functionality would overlap too heavily with prime towers to be practical to not merge them. Take the following examples.

- I am using dual extrusion on a part with some thin cross sections, it would be foolish to have both a prime tower (or 2) _and_ a cooling tower. Why waste material when I can just slow down the prime tower(s) to meet my minimum layer time.

- What if I had a model that is thicker at the top then the bottom, or an hour glass shape, Just like the prime tower, if every layer above the current one needs no assistance to meet the minimum layer time, the tower can stop there. (So the behavior would change from last extruder change to last extruder change _**or**_ last layer < minimal layer time)
[image]

> And it only primes the extruders that are going to be used on the next layer (aside from the structural extruder used for the prime tower outer shell). 

That seems fine, it will then only prime the one extruder. You can also reduce the default minimum volume if you know that its only going to be used for wiping. 

> We could strip out all of those behaviors if you're using it for minimum layer time, but it's a bit clunky and shouldn't really be called a prime tower any more at that point.

Well to be fair, you can prime without a tower, but if you want to prevent stringing you need to wipe the nozzle on something, same problem here. Does it make sense to just move prime tower settings to their own category and rename them "Wipe Towers"?

To recap:
**G1**: Increase layer time to allow each layer to cool for longer.
**G2**: Increase print speed to move nozzle away from smaller cross sections faster.
**G3**: Maintain constant print within model speed to encourage consistency.

Honestly here is my sloppy semi-modular pseudocode solution. Sorry about the many API inconsistencies, still learning the engine.
```c++
//Attempt to reduce the speed via tweaking these individually in this order.
[... cut ...]
                     double inv_factor = 1.0/factor;

                     for (GCodePath& path : paths){
                          if (path.config.getPrintFeatureType() == ftype){
                                  path.speed *= factor;
                                  path.estimates.extrude_time *= inv_factor;
                          }
                     }
                     totalTime = (featureTime * inv_factor) + timeNotInFeature;
             }
          }
          totalPrintTime = totalTime;
    }
}
```

Ghostkeeper, 2020-10-02:
> For the rest of this I am going to reference these goals as **G1**, **G2**, and **G3** respectively.

Ah, g-codes! I know this! :rofl:

Ghostkeeper, 2020-10-05:
We've discussed this with the team but would like to reject this feature request for several reasons:
* Allowing the prime tower to be re-used for different functions than just priming requires some re-thinking of how the prime tower is generated. In particular, it's a matter of when the prime tower is being generated in the slicing process. If it needs to depend on the time estimates, the prime tower needs to be generated after the time estimates have been generated, and also cannot influence the time estimates unless it is seen as a post-processing step, but currently the placement of the prime tower also determines the order or polygons (through minimising travel lengths) so this becomes complicated very fast. It's more or less going to require executing the planning stage twice in order to adjust the height of the tower, which would impact performance.
* There are easy workarounds which work well with well-tuned printers (Lift Head). Contrary to Lift Head, while printing the tower your print head will be too far from the model for the fans to cool down the print so the minimum layer time needs to be different.
* There are other workarounds which work well with badly tuned printers (adding an extra mesh and manually tuning the layer time).

Sorry, but we don't think it's a good idea to add this.

closed 2020-10-05 by Ghostkeeper

Mike111177, 2020-10-09:
Fair enough, worth a shot