Ultimaker/CuraEngine #1338: Couldn't find definition file with ID: deltacomb_base_extruder_0

Li-rr, 2020-10-05:
Hi, I try to use curaengine slice a sample stl by deltacomb_dc20dual.def.json , but i met some problems.

I run this code:
```shell
CuraEngine slice -v -j ~/workspace/Cura/resources/definitions/deltacomb_dc20dual.def.json -o "~/test.gcode" -l "~/chuizi.stl"
```
this is my PATH:
```shell
▶ $PATH

zsh: no such file or directory: /home/lrr/workspace/Cura/resources/extruders:/home/lrr/workspace/Cura/resources/definitions:/home/lrr/workspace/Cura/resources/definitions:/home/lrr/anaconda3/condabin:/home/lrr/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/cuda-10.1/bin:/usr/local/cuda-10.1/bin
```

```shell
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
[ERROR] Couldn't find definition file with ID: deltacomb_base_extruder_0
[ERROR] Couldn't open JSON file: 
[ERROR] Couldn't find definition file with ID: deltacomb_base_extruder_1
[ERROR] Couldn't open JSON file: 
[ERROR] Couldn't find definition file with ID: deltacomb_base_extruder_2
[ERROR] Couldn't open JSON file: 
[ERROR] Couldn't find definition file with ID: deltacomb_base_extruder_3
[ERROR] Couldn't open JSON file: 
[WARNING] JSON setting speed_topbottom has no default_value!
[WARNING] JSON setting speed_wall_0 has no default_value!
[WARNING] JSON setting speed_layer_0 has no default_value!
[WARNING] JSON setting speed_travel has no default_value!
[WARNING] JSON setting speed_travel_layer_0 has no default_value!
[WARNING] JSON setting speed_z_hop has no default_value!
[WARNING] JSON setting acceleration_enabled has no default_value!
[WARNING] JSON setting acceleration_travel has no default_value!
[WARNING] JSON setting acceleration_print has no default_value!
[WARNING] JSON setting acceleration_wall has no default_value!
[WARNING] JSON setting acceleration_wall_0 has no default_value!
[WARNING] JSON setting acceleration_topbottom has no default_value!
[WARNING] JSON setting acceleration_layer_0 has no default_value!
[WARNING] JSON setting acceleration_prime_tower has no default_value!
[WARNING] JSON setting acceleration_support has no default_value!
[WARNING] JSON setting acceleration_support_interface has no default_value!
[WARNING] JSON setting jerk_enabled has no default_value!
[WARNING] JSON setting jerk_print has no default_value!
[WARNING] JSON setting jerk_infill has no default_value!
[WARNING] JSON setting jerk_travel has no default_value!
[WARNING] JSON setting travel_avoid_distance has no default_value!
[WARNING] JSON setting roofing_layer_count has no default_value!
[WARNING] JSON setting roofing_line_width has no default_value!
[WARNING] JSON setting infill_pattern has no default_value!
[WARNING] JSON setting support_z_distance has no default_value!
[WARNING] JSON setting support_bottom_distance has no default_value!
[WARNING] JSON setting support_bottom_enable has no default_value!
[WARNING] JSON setting skirt_brim_speed has no default_value!
[WARNING] JSON setting brim_width has no default_value!
[WARNING] JSON setting prime_tower_size has no default_value!
[WARNING] JSON setting prime_tower_position_x has no default_value!
[WARNING] JSON setting prime_tower_position_y has no default_value!
[ERROR] Couldn't find definition file with ID: deltacomb_dc20dual_extruder_0
[ERROR] Couldn't open JSON file: 
[ERROR] Couldn't find definition file with ID: deltacomb_dc20dual_extruder_1
[ERROR] Couldn't open JSON file: 
[ERROR] Failed to open ~/test.gcode for output.

```

konskarm, 2020-10-05:
Did you try setting the CURA_ENGINE_SEARCH_PATH as it is mentioned in https://github.com/Ultimaker/CuraEngine#running ?

Li-rr, 2020-10-05:
> Did you try setting the CURA_ENGINE_SEARCH_PATH as it is mentioned in https://github.com/Ultimaker/CuraEngine#running ?

Yes, I try, but it didn't work.

```shell
120 export http_proxy=http://127.0.0.1:12333
121 export https_proxy=http://127.0.0.1:12333
122 export CURA_ENGINE_SEARCH_PATH=~/workspace/Cura/resources/definitions
123 export PATH=$CURA_ENGINE_SEARCH_PATH:$PATH
124 #export PATH=/home/lrr/workspace/Cura/resources/definitions:$PATH
125 #export PATH=/home/lrr/workspace/Cura/resources/extruders:$PATH  
```

Li-rr, 2020-10-07:
> Did you try setting the CURA_ENGINE_SEARCH_PATH as it is mentioned in https://github.com/Ultimaker/CuraEngine#running ?

Thank you help, i fix this problem. 

It's my environment variable setting problem.

Setting this, it's work:
```shell
122 export CURA_ENGINE_SEARCH_PATH=/home/lrr/workspace/Cura/resources/definitions:/home/lrr/workspace/Cura/resources/extruders
123 export PATH=$CURA_ENGINE_SEARCH_PATH:$PATH

```

closed 2020-10-07 by Li-rr