Ultimaker/CuraEngine #1339: How to disable auto center

eskeyaar, 2020-10-07:
Hi,
How can I disable the model laying in the center of the build when importing single stl? Also any arguments for doing in CLI too? 

Thanks

nallath, 2020-10-08:
But if it's not putting it in the center, where should it put it? STL doesn't really define an origin (or well, it does define one, but not where that origin is wrt a 3D printer).

eskeyaar, 2020-10-08:
Good question and i think it should put stl where it was saved originally. Because of this feature enabled, i can't load the file in the same place which is previously saved at other software ( slicers). exporting the model as 3mf  and opening the same 3mf file works as it positions the model in save location but not STL. Since STL is only supported for CLI, (make .3mf for CLI too? ) can't use that model to slice them. Other slicers have options to disable auto center, and still, STL loads where it was originally saved. why not cura? or at least in CLI with some arguments to pass for disabling them?

nallath, 2020-10-08:
But what does "saved originally" mean? 3MF explicitly defines the origin to be the front left corner. 

What should be done with STL? Center? Left corner? Every program will use a slightly different one. 

The main reason is that we do't do this for the CLI is because we only use that as a debug interface (and as such, couldn't really be bothered to implement the centering).

eskeyaar, 2020-10-15:
Well i think it should use its place according to its origin. 
Here are some screenshots, originally STL was moved at some X Y Distance based on front left origin of Other Software( PS),
When importing to S3D( to different machines build volume and origin ) , it was still respecting its original X Y Distance on STL was saved. 

Orignal Model saved: 

[image]

S3D Ultimaker 2 + build Volume:
[image]

S3D Finder , Origin is middle of the bed
[image]

Same model, in Cura
[image]

STL Model :
[UM2_cube-40-40-40.zip](https://github.com/Ultimaker/CuraEngine/files/5386260/UM2_cube-40-40-40.zip)

Yes, agree on the CLI for debugging, but the option to uncheck to load center will definitely support other developments. 
it can be a simple check where users can control the placement of STL.
I would love to have this feature or any special arguments for CLI, or any other way to do as other software does.

Thanks

Ghostkeeper, 2020-10-19:
> S3D Finder , Origin is middle of the bed
> [image]

This is a good example of why this feature would be highly confusing. It will also not work well with circular build plates.
I don't really see a good reason why you would want it to load the model in that specific spot on the build plate. To make multiple objects fit together is the only real reason I can think of, but Cura has the merge function for that. The use case is extremely small, but could confuse users that didn't want it or didn't understand the preference if it needs to be enabled first.

To answer your original question: In CuraEngine's CLI, provide the `-s center_object=false` parameter to disable auto centring.

closed 2020-10-19 by Ghostkeeper

eskeyaar, 2020-10-19:
> This is a good example of why this feature would be highly confusing. It will also not work well with circular build plates.
> I don't really see a good reason why you would want it to load the model [...]

It works! Thanks @Ghostkeeper

eskeyaar, 2020-11-04:
@Ghostkeeper is there any special way to do this when using --connect socket method? i tried adding -s center_object=false, but still file was at center.

Ghostkeeper, 2020-11-10:
It shouldn't make a difference. Of course, Cura's front-end will automatically arrange - and therefore centre - any models you load in. This changes the coordinates sent to CuraEngine. This may be what you're seeing. The only way to prevent this is by loading .3mf files instead of any other file type. The 3MF file format specifies that all coordinates are relative to the front-left corner of the printer and shouldn't be rearranged.

Note that `machine_center_is_zero` also causes CuraEngine to add half the machine size to the X and Y coordinates.

eskeyaar, 2020-11-22:
Thanks. 
Jus to confirm we cant use 3mf file with CLI?

Ghostkeeper, 2020-11-26:
No, CuraEngine has no implementation to read 3MF files.