Ultimaker/CuraEngine #1340: Potential Opportunities using Cross-sectional Structural Analysis

Mike111177, 2020-10-09:
So I print a lot of D&D minis. And there is a lot of problems I run into when trying to print small cross sections, (like swords, legs etc). Often times as these thinner and longer parts get printed, the model will flex as the nozzle drags across the top, causing print inaccuracies and sometimes even breakage. I am sure that this would be applicable to many situations.

So admittedly this is idea is still in the "shower thoughts" stage of fleshed out. But I recently read a paper called [Cross-sectional Structural Analysis for 3D Printing Optimization](https://d2f99xq7vri1nk.cloudfront.net/2013_siggatb_CrossSectionalAnalysis.pdf), in which they developed a fast method to identify weak parts of a model as well as parts above weak zones during the print process. They they used this to simply choose the best orientation to print parts in for maximum strength, however I think there is greater potential using this analyses to help with parts with flimsy cross sections.

This method of analyses can find the following information about the model.
- Cross sections in the model that are weaker due to their shape/size.
- The directionality of min/max strength of each cross section.
- Risky print areas that are only supported by weaker cross sections on lower layers, in which the printhead would create greater bending moments.
- Sections that are stronger due to being supported by multiple, weaker cross sections on lower layers.

This information could be used to aid parts that may bend or break during the print process in the following ways.
- Give user suggestions for the best print orientation (Just as they did in the paper).
- Add extra support near and above flimsier parts of the model (see #1239). 
- Since on smaller cross sections we know the strong direction, we could change the infill to be oriented in the direction of maximum strength, that way the nozzle spends less time pushing the face in its weaker axis, reducing bending.
- We could also detect weaker zones and reducing the nozzle speed when moving along the weaker axis to reduce drag force. For example a long rectangle would reduce the nozzle speed when going the short way, but could go normal speed when going the long way.
- Dynamically increase infill in weaker cross sections to improve strength in those areas (#1134) (This admittedly may have limited effect since on these small cross sections walls probably dominate the strength anyway. Also cubic subdivision probably negates the need entirely.)

I am working on a pseudocode to do the analysis, if I can figure out what stage it could run in and how to get the cross section data from CuraEngine I need I might be able to make it a little integrated. But I am curious on any thoughts you guys might have about this before I get too down into the weeds.

Ghostkeeper, 2020-10-12:
Please also reference this documentation providing an overview of the slicing process: https://github.com/Ultimaker/CuraEngine/blob/master/docs/pipeline.md

Once the slicing stage is completed, the cross sections are stored in `SliceMeshStorage::layers`. Once the "generating areas" stage is completed, the rest of those fields there are also filled in (e.g. which parts of the layer are skin).

I think that structural analysis for the 3D prints to optimise orientation is a very good idea. Aside from optimising strength, there is of course also the matter of minimising the need for support, which is an orthogonal problem.

I think that you won't get a whole lot of extra strength by aligning the direction of infill or skin lines to maximise strength. Chances are that there are more than one of these directions for different parts of the model, but to break up the infill area into different areas of different orientation would mean that the infill lines need to be interrupted. This interruption would cause a large fault line in the model that breaks very easily. Orienting the whole model or rotating the "infill line directions" setting is more promising.

There are some services that already try to optimise these things. Fusion360 has one built-in, I think. Topology optimisation is the keyword. It's generally a pretty hard problem because you need to work with 3D volumes built from a triangular surface mesh. If the surface mesh is imperfect in any way the software quickly breaks down.

Mike111177, 2020-10-12:
> Once the slicing stage is completed, the cross sections are stored in `SliceMeshStorage::layers`. Once the "generating areas" stage is completed, the rest of those fields there are also filled in (e.g. which parts of the layer are skin).

If we are going to do things like adjust support areas, change infil etc... then I think this analysis would have to sit in `FffPolygonGenerator::generateAreas` after `sliceModel` runs but before `slices2polygons`. 
> I think that you won't get a whole lot of extra strength by aligning the direction of infill or skin lines to maximize strength.

I agree, that wouldn't be the point of changing infill direction. In-fact it will actually probably make the part weaker. The goal of it is to try to apply nozzle drag in the strongest direction **_during the print_** so hopefully the area flexes less causing less print inaccuracies. 
[image]

### Algorithm Ideas: 
I think that it can be done in a few stages.

#### Stage 1: Coarse rigidity analysis.
Here we iterate through a sample of the layers generated (for example, like every 5th layer or something to cut out work), and we work out for each island (LayerPart) an estimated second moment of area vector and a centroid. This can be done completely parallel. If at then end we find that no island we calculated has a second moment area below a certain threshold, we can skip the rest since the part will probably be fine. Keep in mind, second moment area is actually a vector, so we will be taking into account the direction of weakness.

#### Stage 2: Locate Lowest Weak layer.

Out of all of the layers we have found below our threshold, identify the lowest one (by z height), and then use gradient decent to find the actual weakest layer near it.

#### Stage 3: Build tree and finish evaluating strength.
We now need to do the rigidity analysis for _every_ layer between our lowest weak layer and the top layer.
Simultaneously, we need to build a connectivity tree so we know what layers are supporting layers above them.
Both of these are very parallel processes. We also need to identify "root" nodes/layers, which consist of any layerparts that are either on our lowest weakest layer, or are not supported by anything beneath them (anything that droops down).

#### Stage 4: Climb tree to identify both weak and risky zones.
So to clarify a weak zone is any zone with a low second moment of area in either direction. A risky zone is any zone supported by a weak zone. The farther the z distance between a surface and a weak zone, the more bending moment the nozzle applies to that weak zone. This generally means that you want to try to strengthen the area _between_ the weak zone and the risk zone and also perhaps try to reduce nozzle drag in risk zones.
[image]
If there is a weak zone on a top layer, it doesn't matter because there will be nothing printed above it to have leverage on it.

This also has the caveat that when two weak zones merge at a higher layer, the risk substantially reduces due the strength of the two weak zones being combined. In-fact, the way second moment of area is calculated, the combined strength of to weaker zones can be orders of magnitude greater than just the sum of the two. for example
[image]