Ultimaker/CuraEngine #1343: Retract at layer change generates useless retract after Z move rather than before

richfelker, 2020-10-18:
Enabling retract at layer change should retract the filament before the Z move, so that the Z move does not drag filament out. Instead it generates a useless retract-unretract cycle *after* the Z move.

richfelker, 2020-10-18:
The root cause seems to be the logic in `LayerPlanBuffer::addConnectingTravelMove` to set `final_travel_z` for the previous layer, which causes the Z move to be merged into the end of the previous layer rather than at the start of the new layer. Commenting out that line fixes it. I'm not sure if it breaks anything else.

Ghostkeeper, 2020-10-20:
I'm not seeing a retraction after the Z move. Do you have a Cura project file that reproduces your issue?

Here is a snippet of g-code that I'm getting (with my comments):
```
G1 X121.37 Y102.675 E54.07435        ;Last two extrusion moves.
G1 X121.37 Y112.325 E54.12729
M204 S5000
M205 X30 Y30
G0 F15000 X121.17 Y112.325           ;Move away from outer wall before retracting.
G0 X120.985 Y111.94
G0 X120.745 Y111.7
G1 F1500 E47.62729                   ;Retraction (6.5mm)
;MESH:NONMESH
G0 F600 X120.745 Y111.7 Z3.27        ;Z move!
G0 F15000 X120.385 Y111.34           ;Travel to starting point of next layer (still retracted)
;TIME_ELAPSED:313.300468
;LAYER:30
M204 S4000
M205 X25 Y25
;TYPE:FILL
;MESH:cube.stl
G1 F1500 E54.12729                   ;Unretraction
G1 F4128 X120.383 Y106.471 E54.15935
G1 X112.704 Y108.528 E54.21168
```

richfelker, 2020-10-20:
OK, let me see if I can narrow down the cause and get a minimal testcase to reproduce it.

richfelker, 2020-10-22:
OK, using the model file I originally hit this with, I can reproduce it with nothing but `fdmprinter.def.json` and the following:

- `initial_bottom_layers=0`
- `top_layers=0`
- `wall_line_count=1`
- `infill_line_distance=0`
- `retract_at_layer_change=true`
- `retraction_combing_max_distance=0`
- `wall_0_wipe_distance=0`

Here's a segment of the output gcode:

```
G1 X59.62 Y67.316 E59.04096
G1 X59.01 Y67.641 E59.05397
G1 X58.382 Y67.947 E59.06711
G0 F3600 X58.2 Y68.028
G0 X58.04 Y67.662
;MESH:NONMESH
G0 F600 X58.04 Y67.662 Z0.4
G0 F3600 X58.405 Y67.947
;TIME_ELAPSED:108.642604
;LAYER:1
G1 F1500 E52.56711
M106 S255
;TYPE:WALL-OUTER
;MESH:lin_advance_tower.stl
G1 F1500 E59.06711
G1 F1496.3 X57.775 Y68.228 E59.07143
G1 X57.14 Y68.485 E59.07573
G1 X56.49 Y68.721 E59.08006
```

Adding `retraction_min_travel=0` makes it go away. So the root cause seems to have been that the total combined travel, including the Z move, was less than `retraction_min_travel` and thus caused the retraction around it to be elided, and a separate spurious retract/unretract cycle to be added after the Z move. I still believe this is incorrect though; `retract_at_layer_change` should force the retraction to take place across the Z move at least rather than after it.

Here is the file if you want to reproduce it: 
[lin_advance_tower.stl.txt](https://github.com/Ultimaker/CuraEngine/files/5420275/lin_advance_tower.stl.txt)

(To clarify the name, it's a linear advance calibration test tower - you slice it with overhang speed adjustment to get alternation of speeds and tune K until there's no visible difference in extrusion.)

Ghostkeeper, 2020-10-22:
Hmm, I'm getting something different (comments added myself)...

```
G1 X57.121 Y67.315 E59.04205
G1 X56.511 Y67.64 E59.05505
G1 X55.883 Y67.946 E59.06819 ;Last printed line.
G0 F3600 X55.714 Y67.583 ;Travel away from outer wall.
G1 F1500 E52.56819 ;Retraction!
;MESH:NONMESH
G0 F600 X55.714 Y67.583 Z0.4 ;Z move.
G0 F3600 X55.908 Y67.946 ;Travel to start of next layer.
;TIME_ELAPSED:109.202565
;LAYER:1
M106 S255
;TYPE:WALL-OUTER
;MESH:lin_advance_tower.stl
G1 F1500 E59.06819 ;Unretraction.
G1 F1495.5 X55.278 Y68.227 E59.07251 ;Start printing again.
G1 X54.643 Y68.484 E59.07681
G1 X53.993 Y68.72 E59.08115
```

That is with the following project file via the front-end: [no_useless_retract_in_z_move.zip](https://github.com/Ultimaker/CuraEngine/files/5422869/no_useless_retract_in_z_move.zip)

I also tried with the CuraEngine debug CLI interface, as you seem to be doing: `/home/trin/Gedeeld/Projects/CuraEngine/build-linux/CuraEngine slice -j fdmprinter.def.json -s initial_bottom_layers=0 -s top_layers=0 -s wall_line_count=1 -s infill_line_distance=0 -s retract_at_layer_change=true -s retraction_combing_max_distance=0 -s wall_0_wipe_distance=0 -l /home/trin/Desktop/lin_advance_tower.stl -o test.gcode`

But the result is more or less the same:

```
G1 X59.646 Y67.315 E59.83948
G1 X59.036 Y67.64 E59.84381
G1 X58.408 Y67.946 E59.84819 ;Last printed line.
G0 F5400 X58.239 Y67.583 ;Travel away from outer wall.
G1 F1500 E53.34819 ;Retraction!
;MESH:NONMESH
G0 F600 X58.239 Y67.583 Z0.5 ;Z move.
G0 F5400 X67.904 Y58.315 ;Travel to start of next layer.
G0 X68.005 Y58.36
;TIME_ELAPSED:114.980103
;LAYER:2
;TYPE:WALL-OUTER
;MESH:/home/trin/Desktop/lin_advance_tower.stl
G1 F1500 E59.84819 ;Unretraction!
G1 F1528 X67.702 Y58.982 E59.85253 ;Continue printing.
G1 X67.374 Y59.598 E59.85691
G1 X67.032 Y60.191 E59.8612
```

Indeed in our source code the retraction is forced if Retract At Layer Change is enabled:

https://github.com/Ultimaker/CuraEngine/blob/4702b76e54331cfa355f470c0d541515b4b1da90/src/LayerPlanBuffer.cpp#L101

richfelker, 2020-10-22:
Ah, the difference you're seeing is due to the fix in 96e4c7ca72. So it seems like this is mostly fixed upstream, but I'm still concerned about that last G0 before the retract.

richfelker, 2020-10-22:
OK, the extra G0 seems to be from `moveInsideCombBoundary`, which has a comment that it's making an explicit barrier against moving the retraction before it. This is probably the right thing to do in any case except a single-wall-only, no-infill print, and can be suppressed by turning off combing. I'll check what happens with a model that's actually hollow, though.

richfelker, 2020-10-22:
Uhg, that case is broken - it does an unretracted move outside of the model before retracting and going to the next layer. Apparently the logic in `moveInsideCombBoundary` for ensuring that the destination of the move (after moving "inward") is inside the comb region is wrong.

Ghostkeeper, 2020-10-23:
> Are you sure you're starting with the baseline config?

Pretty sure in both of my tests. Custom FFF Printer doesn't override any settings from fdmprinter. The only real difference then is in the positioning of the model, since the implementation of centering the model in the front-end is different. And in the other test I was explicitly calling with `-j fdmprinter.def.json`. In my CLI test I did correct one of your settings: `wall_0_wipe_distance` is actually `wall_0_wipe_dist`. I don't know why there was a difference in Z of the second layer.

> There shouldn't be an unretracted move before the retraction. Do you know what's causing it?

As you found out in the meanwhile, this is on purpose. If you retract on the outer wall, it'll stop the nozzle on the outer wall and then leave a blob there. That tiny travel move is intended to immediately move away from the outer wall so that there is no blob there. Indeed this makes the assumption that the place you're moving to is not visible, which is not the case if you have 1 wall and no top/bottom skin.

> Apparently the logic in moveInsideCombBoundary for ensuring that the destination of the move (after moving "inward") is inside the comb region is wrong.

Is it not moving inside properly for you? In your g-code snippet I can see the coordinates of a move but it's hard to tell if that is a move towards the inside of the model.

richfelker, 2020-10-23:
> Is it not moving inside properly for you? In your g-code snippet I can see the coordinates of a move but it's hard to tell if that is a move towards the inside of the model.

In that case it is towards the inside of the model, which should be infill. However it also happens, and goes in the other direction (outward) with the model changed to be explicitly hollow, presumably because it just happened to pick the "inner" path as the one whose normal it would move in the negative direction along.

Further, it looks like when components are sufficiently thin that there's no interior apart from the outside wall, combing is allowed in the full region out to the outside of the single wall extrusion, rather than just up to the outline inset by half the line width. This results in the combing taking place with the nozzle positioned halfway over the extrusion and halfway over thin air, rather than along the center of the extrusion, which seems to mar the surface badly. (In practice it's not bad with PLA, but I'm presently doing this tuning with TPU, and I'm pretty sure it would be really bad with PETG too.) Perhaps this should be opened as a separate issue now since it seems independent of the one I reported here, which is already fixed.

Ghostkeeper, 2020-10-27:
> Further, it looks like when components are sufficiently thin that there's no interior apart from the outside wall, combing is allowed in the full region out to the outside of the single wall extrusion, rather than just up to the outline inset by half the line width.

Expected behaviour is that if Combing Mode is set to "all", it can comb within the centreline of the first inner wall in places where there is an inner wall, or within the centreline of the outer wall if there is not. As far as I can see this happens correctly.
[image]
The Maximum Travel Resolution setting may still cause it to go outside of its combing boundary though.
But yes, that should be a separate ticket with its own reproduction steps and such. I'll close this ticket as being fixed.

closed 2020-10-27 by Ghostkeeper