Ultimaker/CuraEngine #1349: Initial extruder other than T0 causes an unnecessary switch N2

DIMI-PIMI69, 2020-10-29:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

DIMI-PIMI69, 2020-10-29:
Hello everyone This is far from the first question, but to which no answer was given.
I have a tough question for you.
the following commands are automatically generated in the slicer:

;FLAVOR:RepRap
;TIME:1412
;Filament used: 0.429382m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.2.0
M141 S50
**T0
M190 S20
M104 S45
M104 T1 S45 ;Fixed T1 temperature
M109 S45
M109 T1 S45 ;Fixed T1 temperature**
M82 ;absolute extrusion mode
G28 ;Home

T2      ;enable duplication mode
M83 ;relative extrusion mode
;LAYER_COUNT:101
;LAYER:0
M107
G0 F3000 X-44.917 Y10.233 Z0.2
;TYPE:SKIRT
G28 ;Home
These commands confuse my 3D printer as they add unnecessary t0 at the beginning. 

Can anyone tell me how I can edit these commands or remove them altogether?

Ghostkeeper, 2020-11-02:
Duplicate of https://github.com/Ultimaker/Cura/issues/7395 .

closed 2020-11-02 by Ghostkeeper