Ultimaker/CuraEngine #1351: Feature request: Vary Infill density by layer

scheerjacob, 2020-11-02:
Hi, 

Currently, there is only one variable you can insert for density in Cura. That density parameter is read by the source code as line_distance. I would like to know how to implement the following request. I would like to vary the infill percentage layer by layer similar to the way infill angle can be adjusted layer by layer using a matrix. The following matrix [0,90] in the infill line_angle parameter would result in an infill pattern where every layer is perpendicular to the one below it. 

For example, What I want to be able to do is input a matrix in the density parameter (like [20, 40] ) and get an infill percentage of 20% for the first layer, 40% for the second layer, and so on until the build is complete. This would be very helpful for an application I am working on where each layers infill impacts the function of the device.

I am sort of a noob when it comes to programming, so I can't understand the source code well enough to implement it on my own. I think I'd need to edit the infill.cpp file in the class addLineInfill or something like that. The variable that needs to be adjusted layer wise is line_distance, but I don't really know how to do more than that.

Any thoughts or ideas for how to do that or at least where to get started?

nallath, 2020-11-03:
What you can already do is use infill modifiers to hack together what you want. See https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura for more information.

But could you also tell us why you want to change it for every layer? What problem do you think that this will solve? There are quite some caveats to changing it every layer; Calculating where the infill is actually supported by the previous infill is going to be a bit tricky. This is also why the gradual infill that we have now works the way that it does now (it halves the density; that way at least a part of the upper layers are supported)

scheerjacob, 2020-11-09:
Hi Nallath,

I am aware of the per model infill settings that can be applied to certain segments of the model.

What I am trying to accomplish is to achieve an infill grid where the two layers are not the same infill density. For instance a 90 degree grid where one layer has an infill density of 50% and the second, perpendicular layer has an infill percentage of 80%.

It has to do with a project I am working on where the infill is the most important characteristic of the build.

Another example would be when changing the infill angle layer by layer. Let's say that I have a matrix that controls the infill angle and it looks like so [0,45,90,135]. However, I want the verticies to all line up. In order to do this, I would like to adjust the infill density on the 45 and 135 layers by a factor of root(2) = 1.732. Then the lines would be spaced slightly larger apart, but the intersection would take place on the same corners as the 90, 0 angles.

Does this make sense?

Thank you

Ghostkeeper, 2020-11-17:
I get where you're coming from, but from a users' perspective, to make the alternating infill density work well, they'd need to calculate the required density carefully for their infill angles to make it line up, and that is the only real use case of the setting.

We're stuck between a rock and a hard place then. We can't expect the user to solve trigonometry problems every time they want to adjust this setting. And we can't auto-calculate the trigonometry problem for any infill angles either, since there isn't always a solution with infill line distances within the bounds of the printer volume. They'd just see that if they adjust the directions in some ways, the infill disappears from certain layers if we automatically try to line up the crossings.