Ultimaker/CuraEngine #1365: There are no non-empty layers in linux

CirnoGiovanna, 2020-11-26:
**Application Version**

4.8.0

**Platform**

Ubuntu 18.04.5

**Steps to Reproduce**

`./CuraEngine slice -v -j resources/definitions/fdmprinter.def.json -o "output/test.gcode" -e0 -l "../tests/testModel.stl"`

**Actual Results**

Progress: start accomplished in 0.000s
Starting slice...
Slicing model...
Progress: layerparts accomplished in 0.000s
Starting inset+skin...
Layer count: 0
Progress: inset+skin accomplished in 0.000s
Starting support...
Stopping process because there are **no non-empty layers**.
Progress: support accomplished in 0.000s
Starting export...
Multithreading GcodeLayerThreader with 4 threads.
Progress: export accomplished in 0.002s
Total time elapsed  0.00s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:0
;Filament used: 0m
;Layer height: 0.2
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06
End of gcode header.
Print time (s): 0
Print time (hr|min|s): 0h 0m 0s
Filament (mm^3): 0

**Expected results**

Progress: start accomplished in 0.000s
Starting slice...
Slicing model...
slice of mesh took 0.000 seconds
slice make polygons took 0.016 seconds
Progress: slice accomplished in 0.016s
Starting layerparts...
Progress: layerparts accomplished in 0.000s
Starting inset+skin...
Layer count: 238
Progress: inset+skin accomplished in 0.015s
Starting support...
Processing platform adhesion
Progress: support accomplished in 0.016s
Starting export...
Multithreading GcodeLayerThreader with 12 threads.
Progress: export accomplished in 0.516s
Total time elapsed  0.56s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:2955
;Filament used: 0.645987m
;Layer height: 0.1
;MINX:30.2
;MINY:32.2
;MINZ:0.3
;MAXX:69.8
;MAXY:59.8
;MAXZ:24
End of gcode header.
Print time (s): 2955
Print time (hr|min|s): 0h 49m 15s
Filament (mm^3): 4121

**Additional Information**

The model "../tests/testModel.stl" is the original test model in source code of CuraEngine.
I build CuraEngine in Ubuntu but its output is always "no non-empty layers". The strange thing is, the same command works well in the CuraEngine in windows, as shown in **Expected results**.
That windows version CuraEngine is installed along with Ultimaker Cura 4.7.1.
These two versions CuraEngine have almost the same log. The difference is the slicing result and the version information at the log beginning: "Cura_SteamEngine version master" and "Cura_SteamEngine version 4.7.1".

Ghostkeeper, 2020-11-27:
Strange. I'm not seeing the same result here on my Ubuntu 20.10 installation, at least on current Master. Here's what I get:

```
~/Gedeeld/Projects/CuraEngine/build-linux$ ./CuraEngine slice -v -j /home/trin/Gedeeld/Projects/Cura/resources/definitions/fdmprinter.def.json -o "test.gcode" -e0 -l ../tests/testModel.stl 

Cura_SteamEngine version master
Copyright (C) 2020 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
loading '../tests/testModel.stl' took 0.001 seconds
Loaded from disk in 0.221s
[WARNING]  -s mesh_position_z="0" -s wipe_repeat_count="5" -s wipe_brush_pos_x="100" -s wipe_hop_amount="1" -s wipe_hop_enable="true" -s wipe_pause="0" -s wipe_retraction_extra_prime_amount="0" -s clean_between_layers="false" -s bridge_skin_material_flow_3="110" -s bridge_skin_material_flow_2="100" -s bridge_skin_speed_2="25" -s bridge_enable_more_layers="true" -s bridge_fan_speed="100" -s bridge_skin_speed="15" -s bridge_sparse_infill_max_density="0" -s bridge_settings_enabled="false" -s adaptive_layer_height_threshold="0.2" -s adaptive_layer_height_variation_step="0.01" -s adaptive_layer_height_variation="0.1" -s adaptive_layer_height_enabled="false" -s wireframe_roof_drag_along="0.8" -s wireframe_fall_down="0.5" -s wireframe_flow_connection="100" -s wireframe_printspeed_flat="5" -s wireframe_printspeed_bottom="5" -s wireframe_roof_inset="3" -s wireframe_height="3" -s flow_rate_extrusion_offset_factor="100" -s flow_rate_max_extrusion_offset="0" -s magic_fuzzy_skin_outside_only="false" -s magic_fuzzy_skin_enabled="false" -s support_conical_enabled="false" -s cross_infill_density_image="" -s cross_infill_pocket_size="2" -s coasting_speed="90" -s max_extrusion_before_wipe="10" -s coasting_min_volume="0.8" -s coasting_volume="0.064" -s draft_shield_height="10" -s conical_overhang_angle="50" -s draft_shield_height_limitation="full" -s draft_shield_dist="10" -s support_skip_some_zags="false" -s material_flow_dependent_temperature="false" -s bridge_wall_coast="100" -s slicing_tolerance="middle" -s relative_extrusion="false" -s magic_mesh_surface_mode="normal" -s magic_fuzzy_skin_point_dist="0.8" -s mold_roof_height="0.5" -s magic_spiralize="false" -s cutting_mesh="false" -s infill_mesh="false" -s print_sequence="all_at_once" -s meshfix_maximum_resolution="0.5" -s alternate_carve_order="true" -s carve_multiple_volumes="true" -s meshfix_keep_open_polygons="false" -s meshfix_union_all_remove_holes="false" -s meshfix_union_all="true" -s mold_enabled="false" -s switch_extruder_retraction_amount="20" -s wipe_retraction_enable="true" -s ooze_shield_enabled="false" -s prime_tower_enable="false" -s raft_base_fan_speed="0" -s raft_base_jerk="20" -s raft_surface_acceleration="3000" -s raft_surface_speed="20" -s raft_base_line_width="0.8" -s mold_width="5" -s raft_base_thickness="0.3" -s raft_interface_line_spacing="0.9" -s raft_interface_line_width="0.7" -s raft_interface_thickness="0.15" -s raft_surface_line_width="0.4" -s raft_surface_thickness="0.1" -s raft_margin="15" -s wireframe_strategy="compensate" -s wireframe_flow_flat="100" -s skirt_brim_minimal_length="250" -s skirt_gap="3" -s wireframe_roof_outer_delay="0.2" -s skirt_line_count="1" -s prime_blob_enable="false" -s support_meshes_present="false" -s prime_tower_position_y="200" -s support_mesh_drop_down="true" -s bridge_skin_density_2="75" -s support_tower_maximum_supported_diameter="3" -s support_tower_diameter="3" -s support_use_towers="true" -s support_fan_enable="false" -s support_roof_offset="0" -s bridge_wall_speed="15" -s minimum_roof_area="1" -s support_bottom_pattern="concentric" -s bridge_wall_material_flow="50" -s support_bottom_enable="false" -s support_roof_enable="false" -s minimum_support_area="0" -s gradual_support_infill_steps="0" -s support_offset="0" -s support_bottom_stair_step_min_slope="10" -s support_bottom_stair_step_width="5" -s small_feature_speed_factor_0="50" -s prime_tower_size="20" -s support_bottom_stair_step_height="0.3" -s support_xy_overrides_z="z_overrides_xy" -s prime_tower_brim_enable="false" -s support_xy_distance="0.7" -s support_bottom_distance="0.1" -s support_top_distance="0.1" -s support_infill_angles="[ ]" -s wireframe_printspeed_up="5" -s raft_base_acceleration="3000" -s support_pattern="zigzag" -s support_angle="50" -s wireframe_nozzle_clearance="1" -s support_type="everywhere" -s support_tree_branch_diameter_angle="5" -s support_structure="normal" -s support_roof_extruder_nr="0" -s support_extruder_nr_layer_0="0" -s support_infill_extruder_nr="0" -s cool_lift_head="false" -s cool_min_speed="10" -s cool_min_layer_time="5" -s z_seam_x="100" -s raft_interface_fan_speed="0" -s support_tower_roof_angle="65" -s mater [... cut ...]
Progress: support accomplished in 0.206s
Starting export...
Multithreading GcodeLayerThreader with 24 threads.
Progress: export accomplished in 1.272s
Total time elapsed  2.15s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:2968
;Filament used: 0.648864m
;Layer height: 0.1
;MINX:30.2
;MINY:32.2
;MINZ:0.3
;MAXX:69.8
;MAXY:59.8
;MAXZ:24
End of gcode header.
Print time (s): 2968
Print time (hr|min|s): 0h 49m 28s
Filament (mm^3): 4139
```

I regularly test this with a fuzz test and it's been working fine recently.

CirnoGiovanna, 2020-11-27:
Thank you for your reply. Is it possible that I didn't compile CuraEngine correctly?
```
cmake ..
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/***/anaconda3/bin/x86_64-conda_cos6-linux-gnu-cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/***/anaconda3/bin/x86_64-conda_cos6-linux-gnu-c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building with Arcus
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found suitable version "3.13.0", minimum required is "3.0.0") 
CMake Warning (dev) at /usr/local/lib/cmake/Arcus/ArcusConfig.cmake:30 (set):
  implicitly converting '' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Building Stb from source.
-- Created Stb installation at: /home/***/threeD/CuraEngine-4.8.0/build/stb-prefix/src
-- Compile with more optimization flags
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/***/threeD/CuraEngine-4.8.0/build
```
Then,
```
make
Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
[  2%] Linking CXX static library libclipper.a
[  2%] Built target clipper
Scanning dependencies of target stb
[  3%] Creating directories for 'stb'
[  4%] Performing download step (git clone) for 'stb'
正克隆到 'stb'...
注意：正在检出 'd5d052c806eee2ca1f858cb58b2f062d9fa25b90'。

您正处于分离头指针状态。您可以查看、做试验性的修改及提交，并且您可以通过另外
的检出分支操作丢弃在这个状态下所做的任何提交。

如果您想要通过创建分支来保留在此状态下所做的提交，您可以通过在检出命令添加
参数 -b 来实现（现在或稍后）。例如：

  git checkout -b <新分支名>

HEAD 目前位于 d5d052c casting to int to remove ptrdiff_t <-> int comparison warnings
[  5%] No patch step for 'stb'
[  6%] No configure step for 'stb'
[  7%] No build step for 'stb'
[  8%] No install step for 'stb'
[  9%] Completed 'stb'
[  9%] Built target stb
[ 10%] Running cpp protocol buffer compiler on Cura.proto
Scanning dependencies of target _CuraEngine
[ 11%] Building CXX object CMakeFiles/_CuraEngine.dir/src/Application.cpp.o
[ 12%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
[ 13%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ConicalOverhang.cpp.o
[ 15%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ExtruderTrain.cpp.o
[ 16%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffGcodeWriter.cpp.o
[ 17%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffPolygonGenerator.cpp.o
[ 18%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffProcessor.cpp.o
[... 60 lines cut ...]
[ 78%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/Date.cpp.o
[ 79%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/FMatrix4x3.cpp.o
[ 80%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/gettime.cpp.o
[ 81%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/getpath.cpp.o
[ 82%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/LinearAlg2D.cpp.o
[ 83%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/ListPolyIt.cpp.o
[ 84%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/logoutput.cpp.o
[ 86%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/MinimumSpanningTree.cpp.o
[ 87%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/Point3.cpp.o
[ 88%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/PolygonConnector.cpp.o
[ 89%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/PolygonsPointIndex.cpp.o
[ 90%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/PolygonProximityLinker.cpp.o
[ 91%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/polygonUtils.cpp.o
[ 92%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/polygon.cpp.o
[ 93%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/ProximityPointLink.cpp.o
[ 94%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/SVG.cpp.o
[ 95%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/socket.cpp.o
[ 96%] Building CXX object CMakeFiles/_CuraEngine.dir/Cura.pb.cc.o
[ 97%] Linking CXX static library lib_CuraEngine.a
[ 97%] Built target _CuraEngine
Scanning dependencies of target CuraEngine
[ 98%] Building CXX object CMakeFiles/CuraEngine.dir/src/main.cpp.o
[100%] Linking CXX executable CuraEngine
[100%] Built target CuraEngine
```

Ghostkeeper, 2020-11-30:
Looks normal to me. The warning in the Stb library is always there. And the warning in CMake with the cast to string too.

Perhaps it's worth trying to disable OpenMP? Mind that you'll get a bunch of warnings then during compilation that it doesn't know the "omp" pragmas, but those shouldn't cause any issues as per the design of OpenMP.

CirnoGiovanna, 2020-12-02:
I try your suggestion as `cmake -DENABLE_OPENMP=OFF ..`, but it still sliced nothing.
After that, I also refer to #1232 and using `cmake -DENABLE_ARCUS=OFF ..` to build CuraEngine. My purpose is to build CuraEngine from source code and run it. Running it in the command line or GUI is the same to me.
However, CuraEngine still gets the result "_Stopping process because there are no non-empty layers._".
```
 cmake .. -DENABLE_OPENMP=OFF -DENABLE_ARCUS=OFF
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building Stb from source.
-- Created Stb installation at: /home/***/threeD/CuraEngine-4.8.0/build/stb-prefix/src
-- Compile with more optimization flags
-- Configuring done
-- Generating done
-- Build files have been written to: /home/***/threeD/CuraEngine-4.8.0/build
(cura_env) ***@***-Inspiron-5437:~/threeD/CuraEngine-4.8.0/build$ make
Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
[  2%] Linking CXX static library libclipper.a
[  2%] Built target clipper
Scanning dependencies of target stb
[  3%] Creating directories for 'stb'
[  4%] Performing download step (git clone) for 'stb'
正克隆到 'stb'...
注意：正在检出 'd5d052c806eee2ca1f858cb58b2f062d9fa25b90'。

您正处于分离头指针状态。您可以查看、做试验性的修改及提交，并且您可以通过另外
的检出分支操作丢弃在这个状态下所做的任何提交。

如果您想要通过创建分支来保留在此状态下所做的提交，您可以通过在检出命令添加
参数 -b 来实现（现在或稍后）。例如：

  git checkout -b <新分支名>

HEAD 目前位于 d5d052c casting to int to remove ptrdiff_t <-> int comparison warnings
[  5%] No patch step for 'stb'
[  6%] No configure step for 'stb'
[  7%] No build step for 'stb'
[  8%] No install step for 'stb'
[  9%] Completed 'stb'
[  9%] Built target stb
Scanning dependencies of target _CuraEngine
[ 10%] Building CXX object CMakeFiles/_CuraEngine.dir/src/Application.cpp.o
/home/***/threeD/CuraEngine-4.8.0/src/Application.cpp:184:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel
 
/home/***/threeD/CuraEngine-4.8.0/src/Application.cpp:186:0: warning: ignoring #pragma omp master [-Wunknown-pragmas]
 #pragma omp master
 
[ 12%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
[ 13%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ConicalOverhang.cpp.o
[ 14%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ExtruderTrain.cpp.o
[ 15%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffGcodeWriter.cpp.o
In file included from /home/***/threeD/CuraEngine-4.8.0/src/FffGcodeWriter.cpp:12:0:
/home/***/threeD/CuraEngine-4.8.0/src/GcodeLayerThreader.h:126:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
     #pragma omp parallel
 
/home/***/threeD/CuraEngine-4.8.0/src/GcodeLayerThreader.h:148:0: warning: ignoring #pragma omp critical [-Wunknown-pragmas]
     #pragma omp critical
 
/home/***/threeD/CuraEngine-4.8.0/src/GcodeLayerThreader.h:164:0: warning: ignoring #pragma omp critical [-Wunknown-pragmas]
     #pragma omp critical
[... 216 lines cut ...]
Starting support...
Stopping process because there are no non-empty layers.
Progress: support accomplished in 0.000s
Starting export...
Progress: export accomplished in 0.001s
Total time elapsed  0.00s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:0
;Filament used: 0m
;Layer height: 0.1
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06
End of gcode header.
Print time (s): 0
Print time (hr|min|s): 0h 0m 0s
Filament (mm^3): 0
```
I also build CuraEngine in windows 10 by MSVC with the same setting and get the same empty gcode.
It seems like `-DENABLE_ARCUS=OFF` build CuraEngine without protobuf and libarcus.
I have no idea what went wrong.

Ghostkeeper, 2020-12-03:
Perhaps it's not able to access the STL file? Although that should give an error message. This is what I get when the file doesn't exist:

```
[ERROR] Failed to load model: /home/trin/bla.stl. (error number 2)
```

And is the `fdmprinter.def.json` coming from the same version of Cura as where CuraEngine was taken from? As in, both from master at the same time, or both from 4.8 or something?

CirnoGiovanna, 2020-12-05:
Thank you very much.
My CuraEngine is ver. 4.8 but my `fdmprinter.def.json` comes from Cura 4.7.1. When I change CureEngine from 4.8 to 4.7.1, slicing success.

closed 2020-12-05 by CirnoGiovanna

Ghostkeeper, 2020-12-08:
That explains it then. The unit of the shrinkage compensation setting changed which probably made your model tiny.