Ultimaker/CuraEngine #1382: Feature: Adaptive speed

rubydesign, 2020-12-13:
Feature Idea: Adaptive speed according to how recently the nozzle has been over the same part

Problem: I am printing some large lampshades, that have a bottom diameter of 30cm. But at the top the diameter is only 6cm.
So the printer (anycubic predator) can quite nicely do the bottom layers at 100-150 mm/s . But at that speed the plastic does not set before the next layer, (at the top) and so it collapses.

Gui: I would think a "minimum cooling time" could be something that is easiest understood. Say north and south of 1s, this would be the minimum time before the nozzle may go over the same place again. 

Implementation: If "normal" execution would have the nozzle going over the same spot in a shorter time, the whole last layer would be slowed down to the achieve the minimum cooling time.

Example: So in my lamp example i would set the speed to 150mm/s and at the bottom layers the nozzle would take 6.2 seconds for a layer. Then i would set "minimum cooling time" to 1s and when the print goes up, radius decreases and at diameter of 190 it would start decreasing the speed to to always give the plastic 1s to cool. At the top (radius 30mm) the speed would drop to 18m/s.

Benefit: The obvious benefit is reduced printing time while getting good results. Now i am printing the whole thing at 30mm/s whereas i have tried the bottom layers at up to 150 and it works. Might get the printing time to a third with this.

Ps: the shape i print is here https://rubydesign.fi/3d/curved_shade
PPS: i am Fedora 33 with Cura 4.8 AppImage

nallath, 2020-12-14:
So like the "Minimum Layer Time" setting?

rubydesign, 2020-12-14:
Ok, sorry I only started this year. Thought I looked at enough videos and looked through ALL the settings. But must have missed that. 

I'll try that, and close of it's what I meant. 

⁣Sent from TypeApp ​

nallath, 2020-12-14:
No worries :) 

If you want more info, you could also check the settings guide plugin: https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2

closed 2020-12-14 by nallath