Ultimaker/CuraEngine #1385: Dangerous file emitted on write errors such as ENOSPC

richfelker, 2020-12-16:
If writing the output file fails with ENOSPC (or other write errors), CuraEngine leaves the partial file in place without clear indication that it's broken. This file of course lacks the end gcode to turn off the heaters, since it's truncated. I just had a file truncated in the first layer nearly destroy my bed surface by leaving a hot nozzle in place over it.

On write error, CuraEngine should truncate the output file to zero length or delete it so that the potentially unsafe file can't be inadvertently used.

konskarm, 2020-12-17:
Does this happen when you use CuraEngine without Cura? Or did you also observe it when saving GCode files in Cura?

rburema, 2020-12-18:
It might be a good idea to do that, but keep in mind that we can never guarantee that the truncation (or removal) after error actually happens. After all, most such errors happen if the file is already partially written. If an error happens, chances are that any subsequent action taken (on detection of the error) will also fail, still leaving the user with a partial, non zero, file.

@konskarm I don't think anything's in place either way (wether engine or frontend) with regards to truncating/erasing partial files.

Ghostkeeper, 2020-12-22:
We've written a wrapper around CuraEngine that fixes this specific problem. It's the Cura front-end. Indeed that writes to a temporary file and then renames the file.

I don't think we need to re-implement such guarantees for the debug CLI mode of CuraEngine.

richfelker, 2020-12-22:
@Ghostkeeper The CLI is what's useful in automated workflows (including for integrated slicing in OctoPrint, etc.), and it does not even exit with nonzero status or print an error message on failed write! I just tested with `-o /dev/full` and it exited with status 0. I really believe safety against emitting a dangerous file belongs here too, but at the very least errors need to be detectable.

Ghostkeeper, 2020-12-24:
You are using our debug interface in a mission-critical application then. It's okay with me, but I don't think it's appropriate to harden the debug CLI for that purpose. The command line interface is not something we'll really support beyond the purpose of debugging CuraEngine. We're fine with people using it differently but as long as something is not in the way of debugging, I don't think we should sacrifice the time and code maintenance for it. Sorry!

Whatever automation you're using it in will need to implement the safety precautions then. For instance you can use CuraEngine without `-o` parameter to let it output to stdout, and then pipe that output to a (temporary) file yourself in a safer way.

closed 2020-12-24 by Ghostkeeper