Ultimaker/CuraEngine #1386: [ERROR] CMakeFiles/_CuraEngine.dir/build.make:80: *** target pattern contains no '%'.

xypron, 2020-12-20:
**Application Version**
git d509e7e14975e55a0

**Platform**
Debian Bulleye

**Qt**
Debian Bullseye 5.15.2+dfsg-2

**PyQt**
pyqt5-dev (5.15.2+dfsg-1) 

**Steps to Reproduce**
```
mkdir build
cd build
cmake ..
make
```

**Actual Results**
```
make
[  2%] Built target clipper
[  9%] Built target stb
CMakeFiles/_CuraEngine.dir/build.make:80: *** target pattern contains no '%'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:129: CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

The defective line is:
build/CMakeFiles/_CuraEngine.dir/build.make:80
Cura.pb.h: protobuf::protoc

The error relates to the usage of  colons.

**Expected results**
Build success

**Additional information**
* cmake 3.18.4
* GNU Make 4.3

xypron, 2020-12-20:
Cf. #848

xypron, 2020-12-20:
This Debian package has to be installed: 'protobuf-compiler'.

We should add a check in CMakeLists.txt

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0b1bdeac..a8a2093d 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -22,6 +22,10 @@ if(ENABLE_ARCUS)
     find_package(Protobuf 3.0.0 REQUIRED)
     find_package(Arcus REQUIRED)
     add_definitions(-DARCUS)
+    find_program(PROTOC "protoc")
+    if(${PROTOC} STREQUAL "PROTOC-NOTFOUND")
+        message( FATAL_ERROR "Protobuf compiler missing")
+    endif()
 endif()
```

Ghostkeeper, 2020-12-23:
Shouldn't this already ensure presence of Protoc?

    find_package(Protobuf 3.0.0 REQUIRED)

xypron, 2020-12-23:
Am 23. Dezember 2020 17:12:12 MEZ schrieb Ghostkeeper <notifications@github.com>:
>Shouldn't this already ensure presence of Protoc?
>
>    find_package(Protobuf 3.0.0 REQUIRED)

No, this is insufficient. It only checks for the library which in Debian is in a separate package.

PR #1387 (2021-01-08): CMakeLists.txt: check for Protobuf compiler
Fixes: #1386

On Debian there are separate packages libprotobuf-dev and
protobuf-compiler. So when building with ARCUS it is not enough to check
that the library is present. We must also check that the Protobuf compiler
is available.
commit 93ea151e (2021-01-08): CMakeLists.txt: check for Protobuf compiler

Fixes: #1386

On Debian there are separate packages libprotobuf-dev and
protobuf-compiler. So when building with ARCUS it is not enough to check
that the library is present. We must also check that the Protobuf compiler
is available.
closed 2021-01-08 by rburema