Ultimaker/CuraEngine #1388: [4.8] [BUG] CPU Usage Maxed out on slicing particular model (Related to Latest Build Versions)

Trini8ed, 2020-12-21:
4.8 and possibly greater than 4.6 (unsure what versions above 4.6 work at this time)

Windows (Windows 10 19041.685), Linux (Ubuntu 20.04.1 LTS)

N/A

N/A

Linux (Driver Version: 460.27.04    CUDA Version: 11.2)
Windows (Driver Version: 457.30    CUDA Version: N/A)

**Steps to Reproduce**
Use CuraEngine 4.8 and use the test stl file provided in attachments. Run through CLI with the parameters I have entered below.

`slice -j {insert location to definitions here}\prusa_i3_mk3.def.json -s infill_pattern=lines -s infill_line_distance=2.50 -s layer_height=0.20 -s machine_always_write_active_tool=false -s extruder_prime_pos_abs=false -s relative_extrusion=true -s support_structure=normal -s infill_pattern=lines -s mesh_rotation_matrix=[[1.00,0,0], [0,1.00,0], [0,0,1.00]] -l input_file.stl -o output_file.gcode`

Insert the definition path location in the input arguments and run the command. You should now be slicing the model.

STL File:
[Test 2.zip](https://github.com/Ultimaker/CuraEngine/files/5721751/Test.2.zip)

**Actual Results**
Your CPU should max out at 100% CPU usage and take over 5 minutes or longer to slice the model. My windows machine has an i9-7980XE with all cores maxed during this test. My server which has a Tesla M40 and 24 cores running on ubuntu also gets all of its cores capped on this specific STL file. Note I'm only reporting this as a bug due to the fact this exact file took 11 seconds with the same input parameters using CuraEngine 4.6 and not CuraEngine 4.8.

**Expected results**
The model should take more than 20 seconds to render on various hardware instead of the 5 minutes plus on CuraEngine 4.6 I have downgraded the version in my server and local development build to 4.6 and this seemed to have fixed the issue.

**Additional Information**
I have tried removing as many parameters as I could thinking that "meshfix_keep_open_polygons=true" was the main reason for the slow down but that is not the case.

Ghostkeeper, 2020-12-24:
Probably harmless, but there's several unnecessary setting overrides in that command:
* `infill_pattern=lines` is set twice.
* `machine_always_write_active_tool` is set to what is already the default.
* `mesh_rotation_matrix` is set to what is already the default.
* `support_structure` is set to what is already the default. And support is disabled anyway.

That said, I can also reproduce your issue on the debug CLI interface of CuraEngine, like you. I'm unable to reproduce it via the GUI though. So one of the `value` properties for a setting is preventing this issue in Cura, I'm guessing. Trouble is, there are hundreds of those.

Trini8ed, 2020-12-25:
Yes sorry I apologize for the confusion I never used the GUI for Cura, I just use the debug CLI through CuraEngine I just thought it was interesting that the version difference between 4.7 and 4.8. I'll have to tinker around and see what value property was not being set correctly that caused it. Might be easy to figure out the culprit value by comparing the resource definitions default values and see what changed between the two CuraEngine versions. Also I have it on my todo list to migrate over from the debug CLI to the RPC instead, is there any benefit to using the RPC over the Debug CLI other than the isolation factor?

Ghostkeeper, 2020-12-28:
There were no real changes in 4.7 vs. 4.8 in the defaults in fdmprinter. The only meaningful change was in the shrinkage compensation factor setting which changed their units from being 0-oriented to 100%-oriented. That setting has no `value` override though so it'll behave the same when running via CLI.

There is no real difference between using the command line interface over the Protobuf communication channel aside from that the Protobuf connection is supported in production. The slice result is the same, but it means that we don't have to re-implement all of Cura's front-end features in CuraEngine all over again. But if you run it through Cura's front-end that reads the `value` functions from the definitions which changes a bunch of settings. I was pointing out that the problem must be among those.