Ultimaker/CuraEngine #1389: settings.json with human readable list of options that are required/optional per version

andrewhodel, 2020-12-28:
Where is this at?

Ghostkeeper, 2020-12-29:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

closed 2020-12-29 by Ghostkeeper

andrewhodel, 2020-12-29:
Thanks,

Which machine is that for?

andrewhodel, 2020-12-29:
Why not get rid of the slice parameter to curaengine if there will be that
many required options?

I am confused about why you would expect people to type that many options.

Do you have a list of required params?

Andrew

andrewhodel, 2020-12-29:
https://github.com/Ultimaker/CuraEngine/issues/1390

Ghostkeeper, 2020-12-30:
> Which machine is that for?

This is the parent definition of all printers. It's not a particular printer. For particular printers, there are files inheriting from fdmprinter.def.json in the [definitions folder](https://github.com/Ultimaker/Cura/tree/master/resources/definitions).

> Why not get rid of the slice parameter to curaengine if there will be that
many required options?

Because we use the `slice` command for debugging CuraEngine.

> I am confused about why you would expect people to type that many options.

You can use the `-j` option to load in such a JSON file automatically. Not all command-line interfaces are even capable of writing commands long enough to include all settings.

> Do you have a list of required params?

Each of them are required under certain circumstances. The circumstances under which they are required are written in the `enabled` property of these settings (defaulting to "always required" if not present). It's easiest to just always provide all of them, really.

andrewhodel, 2020-12-30:
So are you saying that the -s options override the options in the JSON file
passed to -j?

I didn't read that anywhere in the documentation, also why is there no link
to that file in the documentation?

Andrew

Ghostkeeper, 2021-01-07:
> So are you saying that the -s options override the options in the JSON file
passed to -j?

I didn't say that, did I? It's wrong, so I'm sorry if I did. The parameters are executed in order of appearance, so `-s key=value -j file.json` would make the JSON file override the setting, but `-j file.json -s key=value` would make the `-s` override whatever is in the JSON file.

The command line interface is only a debugging tool. Usage documentation in the `--help` could be improved, yes, but it's not that important and we should be careful to not make it too long of a story in there either. Anyone that is involved enough to try to fix bugs with CuraEngine will know where to find .def.json files, hear it from a colleague or find it rather quickly, since they must also have the front-end then. If you find a better way to write it down, we'll be happy to improve it though, or merge suggested changes.

andrewhodel, 2021-01-07:
Don't you think it makes more sense to always have command line options
override configuration file options?

andrewhodel, 2021-01-07:
Just as a basic, well I am going to execute this command and there is a
file but I am trying out configuration options and effectively it is
ridiculous to expect to go change the configuration file each time.

Ghostkeeper, 2021-01-07:
Yeah myself I pretty much always put the `-j` in front to get all of the defaults from fdmprinter or a certain other printer and then write the `-s` afterwards. It can be useful to use a JSON file as a re-usable profile though which is why it's implemented like this. You can use multiple JSON files and they'll override each other in the order in which you specify.

JSON files are useful specifically because you don't want to change them all the time to test different things. It's useful to have a basic set of defaults or a "profile" of re-usable settings that you don't often change during development. I don't see why you would need to change it all the time if you can use `-s` for specific setting overrides.

andrewhodel, 2021-01-07:
It is a matter of someone not reading the documentation that is currently
sparse anyway, and the default being that if they were testing options and
"accidentally" putting the option for the config file and the option for
the individual option in the wrong order, then it would still work properly
for them.

That of course changes nothing in the grand scheme of things and makes your
software more user friendly which means more people support it, etc, etc,
etc...

andrewhodel, 2021-01-26:
@Ghostkeeper what happened to startCode="escaped string w/ \n" and endCode="escaped string w/ \n"?

andrewhodel, 2021-01-26:
Seems that there should be an option to replace whatever CuraEngine thinks it should put there.

andrewhodel, 2021-01-26:
Nevermind, was in the long settings file nobody wants to read.