Ultimaker/CuraEngine #1392: Bug? Bridged inner walls appear narrower than normal inner walls

Sophist-UK, 2021-01-09:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fvrmr, 2021-01-15:
Can you share your project file? So I can reproduce this.

Sophist-UK, 2021-01-15:
[TEST.zip](https://github.com/Ultimaker/CuraEngine/files/5820988/TEST.zip)
Sorry - I had already zipped it up - I forgot to attach it to the original issue.

Ghostkeeper, 2021-01-18:
It's nice how we can nowadays just load your project files in Cura since the DiscoEasy definition is built-in now :) You've reported a lot of bugs so it makes things a lot easier.

I think what you're seeing is the result of Compensate Wall Overlaps. It's not just happening in the bridging lines, but also in the non-bridging part further out. Compensate Wall Overlaps finds the walls that overlap and reduces the flow rate of one of them (whichever prints last). It doesn't adjust the position of that line though, expecting the material to flow over to fill the gap. So in theory it still partially overlaps with the neighbouring wall and then has an equal-sized gap on the other side. In practice there will not be a gap since the required material is still extruded very close to the gap.

However I can imagine that this doesn't work very well when bridging. There is no counterpressure then to force the material to flow over into the gap.

This problem is resolved in our [experimental version with variable line widths](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha) which completely removed that algorithm in favour of just never creating overlapping walls any more. So once Ultimaker/CuraEngine#1210 is merged, this problem should be removed as well.

Ghostkeeper, 2021-12-14:
This is fixed by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper