Ultimaker/CuraEngine #1396: [4.8.0] Cura generates G1 moves with no extrusion causing stalls.  Precision loss in extrusion moves.

jszymon, 2021-01-24:
**Application Version**
4.8.0

**Platform**
Ubuntu 20.10, Ender 3, Klipper3d

**Steps to Reproduce**
Just slice the attached project (a simple ring)

[CE3_ring.3mf.gz](https://github.com/Ultimaker/CuraEngine/files/5862242/CE3_ring.3mf.gz)

**Actual Results**

The G-code contains extremely short G1 moves without extrusion which cause stalls, e.g.:

G1 X115.238 Y84.983 E57.52195
G1 X114.643 Y85.031 E57.5226
G1 X114.627 Y85.033
G1 X114.025 Y85.09 E57.52359
G1 X113.63 Y85.136 E57.5239

full gcode:
[CE3_test_ring.gcode.gz](https://github.com/Ultimaker/CuraEngine/files/5862244/CE3_test_ring.gcode.gz)

**Expected results**

Commands within walls which do not extrude should be avoided.

**Additional Information**

I traced the problem to the following:

1. Cura emits an extra, very narrow inner wall with very low flow rate

2. The wall has many very short moves (see above)

3. Short distances with low flow rate cause extremely short extruder
moves which require more than 5 decimal digits to represent.  If,
after rounding the extruder move is 0 it is omitted.

4. Even if extruder moves are present the precision is lost with very
short moves and low flow rate which may lead to erratic flow rates in
consequent moves.

Possible solutions:

a) get rid of short moves (4.8.0 was supposed to solve the issue but apparently didn't)

b) increase extruder moves precision to more than 5 digits.  This may
be a good idea even if (a) is implemented.  With very low flow rates
(e.g. during ironing) this could affect extruder performance.

Maybe dynamic precision depending on move length would be a good idea?

c) the extra inner wall seems to overlap an existing one.  Is it even necessary?

fvrmr, 2021-01-26:
Thanks, I will discuss this with the team. Keep you posted!

jszymon, 2021-01-26:
Hi,
Thanks for looking into the issue.

I investigated the issue a little bit more:
1. Managed to compile Cura engine from git and change extruder move precision.
2. Wrote a script which reads the gcode and plots extruder_distance / move_distance for consecutive moves
The plot should be flat over long sequences of moves.

Here is what I got for 5 digits E moves precision
[image]
and for 14 digits:
[image]

The flow is perfectly even in the outer wall but erratic in the inner walls.
Interestingly high precision helped with one outer wall (good!) but had no effect on the really thin one mentioned in the issue.

Could it be that Cura is using this second wall to fill in really tiny gaps somewhere?
Anyway, both inner walls contain a huge number of very short moves...

Ghostkeeper, 2021-01-27:
We discussed this issue today. There are two aspects to this issue that made us decide to not fix it.

Firstly, the wall here is output as a G1 ("print move" according to the documentation) with 0 flow. This is perfectly valid output, and should not be causing issues in your firmware. If it does, we consider that to be a problem with your firmware. Cura can output such moves in 3 cases: If the flow was manually set to 0, when coasting, or when compensating wall overlaps. In your case, you have an inner wall which is almost completely overlapping another inner wall, practically reducing its flow to 0 and you end up with some moves with 0 flow, down to 10 nanometre accuracy. Cura outputs G1 there, not G0, since the move is supposed to be part of a wall, not a travel move. It's just with 0 flow.

Secondly, there is the problem that this move itself is tiny which can cause buffer underruns. That is caused by the Wall Overlap Compensation which breaks up lines into different parts with different amounts of overlap. This is considered a bug as far as I'm concerned; it should still adhere to the Maximum Resolution. However it is solved in #1210 because the Wall Overlap Compensation is removed there.

As a workaround, I suggest you set the Minimum Wall Flow setting to 50% (or really anything above 5%) to prevent creating walls with almost 0 flow. Apparently your printer doesn't deal with those well. But really they are best avoided in every print and it should be the default in my opinion.

closed 2021-01-27 by Ghostkeeper

jszymon, 2021-01-28:
Hi,

thanks for the info.  Agree: the real problem in Wall Overlap Compensation. 

Looking forward to #1210 being merged! Currently I was able to fix the problem using the ArcWelder plugin.

Just FIY: stalls are not caused by buffer overflows.  The problem is uneven moves (in XY and E) which klipper does not concatenate (the author claims it should respect input from the slicer, wchih I agree with) and has to slow down to change extrusion rates so fast. See an issue below (very similar to mine), where low precision causes consequent moves which are too different.to smoothly merge

https://github.com/KevinOConnor/klipper/issues/513#issuecomment-410757812