Ultimaker/CuraEngine #1404: Bridge's specific "Skin expand distance" for better bridge adhesion

Cecchellone, 2021-02-05:
**Application Version**
4.8.0

**Actual Results**
To enhance bridge's attachments adhesion i **must** raise the `Skin Expand Distance` for the whole project, even if I don't want to actually expand the skin's distance. This often results in more fragile prints and a little waste of plastic.

**Expected results**
Being able to separately edit the normal `Skin Expand Distance` but also a hypothetical `Bridge Attachments Expand Distance` separately

**Additional information**
`Bottom Skin Expand Distance` is not enough.

fvrmr, 2021-02-11:
Hi @Cecchellone thank you. I will discuss this with the team. Keep you posted!

fvrmr, 2021-02-12:
Hi @Cecchellone I have discussed it with the team. But I have to defer this request, because bridging is still in experimental mode and there are other important bridging issues that have higher priority.
Hope you understand.