Ultimaker/CuraEngine #1412: Slicing with 0% support density does not terminate if the cross pattern is used.

ThomasRahm, 2021-02-12:
**Application Version**
4.7.1

**Platform**
Windows 10

**Steps to Reproduce**
Set Support Pattern to "cross" and Support Density to 0% with support enabled.

**Actual Results**
The slicing never finishes.

**Expected results**
The support should have no infill.

**Additional Information**
Issue is that support line distance is set to 0 when the support density is 0.
This causes the following loop of SierpinskiFillProvider.cpp (line 73) to never terminate:

```
coord_t aabb_size = min_line_distance;
while (aabb_size < max_side_length)
{
    aabb_size *= 2;
    depth += 2;
}
```
As aabb_size will be 0.

I am not sure how to fix this issue, without potentially adding a new one.

fvrmr, 2021-02-18:
Hi @ThomasRahm thank you for your report.
I can also reproduce this in 4.8. I will discuss it with the team.
Keep you posted!

fvrmr, 2021-02-19:
Hi @ThomasRahm I have discussed it with the team. But I have to defer this. Because it is really exceptional to run into this. So it is not a priority for us to fix. Hope you understand.

lospo, 2022-05-23:
Hi thomas, this is not a bug!
I've talked about your mod, maybe you are interested!
https://community.ultimaker.com/topic/40567-cura-tree-support-mod-ts2/
Have a good day

closed 2024-11-22 by ThomasRahm