Ultimaker/CuraEngine #1424: Gaps in sliced output

daniel-starke, 2021-02-27:
**Application Version**
4.8.0

**Platform**
Windows 7 Professional SP1 x64

**Display Driver**
4.1.0 NVIDIA 445.75

**Steps to Reproduce**
Slice the attached project.
[piece-hole.3mf.zip](https://github.com/Ultimaker/CuraEngine/files/6055373/piece-hole.3mf.zip)

**Actual Results**
Gaps in the model are visible in the preview.
[image]

**Expected results**
The model has real round holes in the middle and no gaps in the wall.

**Additional Information**
This issue did not occur with Cura 3.6.

fvrmr, 2021-03-02:
Hi @daniel-starke thank you for your report.
If I slice your model in our Arachne release, it doesn't happen. 
[image]

Read [here](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha) more about Arachne. 

I will close this issue because it is solved, but feel free to reach out if you have more questions.

closed 2021-03-02 by fvrmr

Ghostkeeper, 2021-03-05:
This is caused by the Maximum Resolution setting. Lowering that to a bit finer resolution fixes the issue.