Ultimaker/CuraEngine #1431: Idea: second level build plate size

Geitde, 2021-03-11:
I have an hopefully interesting idea for cura.

Many printers are able to leave the build plate, when moving the print head. So in theory they could print bigger than the build plate. In the following sections I will refer the out side area as overdrive.

When I currently want to utilize this "overdrive" space I have to increase the build plate, even so the actual print only uses the normal build plate size for the base. Now I have to fiddle around to get the part positioned to match the actual build plate again, within the newly defined one, so the first layer sticks fine and the print can now grow sideways to leave the pysical build plate due to using overhangs.

My idea would be to have an option to define the "printable air space" of a printer and an option to utilize this overdrive area, which is currently not as easy as described above as you need to make sure to manually block supports on such "overdrive" areas, which only can be reached by overhangs anyway.

I easily could increase by print volume by around 40mm on some of my printers, which is often more than enough to avoid splitting a print or print it in some odd orientation. With some smart support structure like tree supports it would even possible to use e.g. 4mm of support below the entire print and then starting the real object. This slicing an object in mid air, gives the support a way to "grow" below the print bed air expansion and build a foundation like an oversized raft.

I know this looks like a feature which would be rarely used, but I imagine many people had an issue with their build plate being just a little to close to fit a model. 

 Thanks for reading this and keep up the good work!

fvrmr, 2021-03-16:
Hi @Geitde thank you for your feature request. 
I will discuss this with the team, but I think the big issue here is the support. 
Keep you posted!

fvrmr, 2021-03-23:
I have discussed it with the team. And we have decided to defer this issue. Because there are a lot of things you we need to take  care of (like support) for developing this feature. 
Hope you understand.