Ultimaker/CuraEngine #1434: Bug - Slicing Fails for special Coasting settings

TimoErdmann, 2021-03-21:
**Application Version**
Cura 4.7.0 to 4.8

**Platform**
Windows 10 

**Steps to Reproduce**
1. enable Coasting
2. set the Coasting Volume to a pretty low Value like: 0,03
3. set the Minimum Volume Before Coasting Value pretty high, like: 10

**Actual Results**
Slicing starts but CuraEngine is crashing. The log says: 
[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [925]: Backend quit with return code 3221225477. Resetting process and socket.

**Expected results**
Slicing results in a file

**Additional Information**
Slicing just stops no further information is given

fvrmr, 2021-03-22:
Hi @TimoErdmann thank you for your bug report. 
I am not able to reproduce this, see here:
[image]

Could you provide me your project file? (file - save project) 
Thank you in advance!

TimoErdmann, 2021-03-22:
Hey fvrmr,
thanks for looking into it. It looks like it depends on other settings too. But the Coasting feature was the most dominant factor for me. Though it changes with the size of the stl and the number of parts. The infill pattern seems to have a effect too. 

i think it has to be a memory or data allocation problem but that is just guessing.

[CFFFP_Farbplattchen11.zip](https://github.com/Ultimaker/CuraEngine/files/6183333/CFFFP_Farbplattchen11.zip)

fvrmr, 2021-03-22:
Thanks! I am able to reproduce it.
I will discuss it with the team. Keep you posted!

closed 2021-03-24 by Ghostkeeper

Ghostkeeper, 2021-03-24:
When investigating this, I found what the issue was and made a fix.

When coasting very small contours, below the minimum coasting volume, it'll linearly decrease the coasting volume based on the fraction of volume extruded divided by the minimum coasting volume. Because you use a fairly low coasting volume and a large minimum coasting volume, the distance it calculated to comb the very tiny letters in this model became less than 1 micron, rounded down to 0.
Then it proceeded to look for which segment the combing needs to stop at. Because of an edge case, it would think it needed to comb until before the first line segment. Because computers are dumb, this underflowed to the 4.2 billionth line segment. Since that is out of range of the memory it crashed.

A workaround until the 4.9 release is to reduce the minimum coasting volume a bit. Like, 9mm³ instead of 10 would probably be okay.

commit a75b9579 (2021-03-24): Fix crash if coasting ends up less than 1 micron

It would then try to coast until the -1st segment. This would then underflow and try to coast until the 4.2-billionth segment. This would segfault.

Fixes #1434.