Ultimaker/CuraEngine #1436: Z Seam Sharpest Corner is numerically unstable

richfelker, 2021-03-31:
Given a model with a large number of Z Seam Sharpest Corner candidates that are all approximately the same sharpness, such as a (in my case skew) cylinder, the seam jumps around from one layer to the next, making warts on the print and making the print take unnecessarily longer.

If there are N candidate points all within some epsilon of being the sharpest, CuraEngine should do something like choosing the one closest to the seam point in the previous layer, or closest to some special point (e.g. the user specified position) rather than choosing one essentially at random, so as to optimize print time and avoid surface quality issues.

fvrmr, 2021-04-02:
Hi @richfelker could you tell me which Cura version you are using?

richfelker, 2021-04-02:
It's from git, Jan 12, 9ead8ee057. If you think this is already fixed in a later version I'm happy to check that.

rburema, 2021-04-06:
Hi @richfelker 

Our QA/testers have recently found similar issues.
Is this comparable to what you're seeing?

[image]

[image]

If so, it's probably related to an internal ticket of ours, CURA-8149, which we hope to fix before 4.9.0-final is out (don't think it's fixed in the beta).

richfelker, 2021-04-06:
Yes, it looks similar, but it looks to me like in both slicings the Z seam is jumping around undesirably, just less so in 4.8. Ideally it would not move at all when there is not a significantly *better* candidate than the closest point.

If it would help you debug the behavior, I could get you a file it happens badly with. My file is a hollow skew cylinder with single-wall, designed for testing/tuning linear advance (by using overhang settings to cause one side of the cylinder to be printed at reduced speed). I expect moving to next layer to be possible without travel, but with default Z-seam at sharpest, there's gratuitous travel jumping all over at layer change.

Ghostkeeper, 2021-04-15:
This is supposed to be fixed since this PR: https://github.com/Ultimaker/CuraEngine/pull/1438 We've tested that with the case that we were seeing (pictured above) and similar models.

Would you like to re-test your case? And are you able to build that yourself?

richfelker, 2021-04-15:
> Would you like to re-test your case? And are you able to build that yourself?

Can confirm it's fixed. Thanks!

closed 2021-04-20 by Ghostkeeper