Ultimaker/CuraEngine #1441: Discontiguous top/bottom fill printing

richfelker, 2021-04-19:
After upgrading from 4.8 to 4.9-beta, I find that top/bottom fill that used to print mostly contiguously now gets split up into multiple gratuitously discontiguous parts with travel between them. Sometimes it even chooses to start in the middle of a feature, going in one direction, while the other direction is entirely a "dead end" with no way to return to it except travel, rather than starting at the "tip" of the dead end.

If needed I can prepare some minimal test cases.

Ghostkeeper, 2021-05-06:
Hmm, yeah I could reproduce something like this. Here is a print of a simple [model](https://github.com/Ghostkeeper/SlicerTestModels/blob/master/my/toothpaste_squeezer.scad) in Cura 4.8:

[image]

After it completed the walls, it travels to one of the corners perpendicular to the angle of the skin lines. This is so that it doesn't start in the middle somewhere, which removes one of the discontinuities of the skin area. That's the purpose of the travel move to a seam-avoiding location here:

https://github.com/Ultimaker/CuraEngine/blob/227584cb2a6c68a0285d35ff11cb4ed6db00629f/src/FffGcodeWriter.cpp#L2520

However in the current master (and probably in 4.9 then) this doesn't work any more. It just seems to start with whichever skin line is closest:

[image]

ky438, 2022-12-08:
@rburema - is there any update on this bug?

rburema, 2022-12-08:
I still see the same part(/project) now being printed (for the fill on top) in 5 parts instead of 3 like it was in 4.8

So, going from the documentation attached to `getSeamAvoidingLocation`, which states that its function (or at least its method of attaining that) is to have as few 'parts' of fill as possible -- this issue is not fixed yet.

It doesn't seem to have an attached (internal) ticket either (linked here nor when searching) and it wasn't closed either, so this might never have been brought up. I'll add it to the list of things to discuss.

richfelker, 2022-12-11:
I haven't followed up on this since it seems that monotonic skin fill avoids it entirely, and is vastly preferable anyway. So I don't have any input on whether it's still happening.