Ultimaker/CuraEngine #1442: Excess travel between outline path components with skin_outline_count>0

richfelker, 2021-04-19:
In a region with holes, the slicer generates repeated travel back and forth between the outer outline and the hole outlines when `skin_outline_count` is nonzero. This is even worse with `outer_inset_first`. For example with a region R and hole H:

1. R wall 0
2. H wall 0
3. H wall 1
4. R wall 1
5. R skin outline 0
6. H skin outline 0

(3 travels) In a worst case, I think 5 and 6 are even sometimes reversed (4 travels). The ideal order would be:

1. R wall 0
2. R wall 1
3. R skin outline 0
4. H wall 0
5. H wall 1
6. H skin outline 0

(only 1 travel)

Or, if "outer inset first" is supposed to imply that all components of the outer inset are done before any inner ones can start:

1. R wall 0
2. H wall 0
3. H wall 1
4. H skin outline 0
5. R wall 1
6. R skin outline 0

(2 travels). However I don't think "outer inset first" is intended to imply such a constraint, since it does not impose one for non-connected components of the layer.

The suboptimal order here increases printing time for top/bottom layers quite a bit, especially on printers with slow retraction, and mars the skin unnecessarily if you allow combing through skin.

richfelker, 2021-04-19:
Implementation-wise, I suspect this could be fixed just by making `skin_outline_count` increase the wall count in the affected regions rather than treating the new outlines as part of the skin. But that might be problematic to do if the skin does not fill the entire outer region, and it would probably change how speed/acceleration profiles apply to the extra outlines.

Ghostkeeper, 2021-05-06:
> For example with a region R and hole H:
> 
> 1. R wall 0
> 2. H wall 0
> 3. H wall 1
> 4. R wall 1
> 5. R skin outline 0
> 6. H skin outline 0

This order is the expected outcome if Optimize Wall Printing Order is disabled. There is another constraint that you didn't take into account here, which is that each of the structures (walls, skin, etc.) must be printed in order. This is to reduce long-term flow changes, which cause overextrusion and underextrusion when the speed, line width or flow changes. Skin is often printed at a different speed and regularly a different line width, so the constraint is that all walls should be printed together and all skin should be printed together. We consider that to be a stronger requirement than the requirement to minimise travels (which don't interrupt the flow rate as much, according to Ultimaker's tests).

> (2 travels). However I don't think "outer inset first" is intended to imply such a constraint, since it does not impose one for non-connected components of the layer.

Indeed, Outer Inset First is only intended to apply to the walls, right now. We could consider changing that but that violates the aforementioned constraint again.

For some shapes, the user can increase the Wall Line Count and reduce the Extra Skin Wall Count equally. However the point of having walls around just the skin and infill is so that there is a border between skin and infill for the skin/infill lines to rest on. So it kind of defeats the point a bit.

However there is another case where some improvement is possible still, because if you have 2 extra skin walls instead of 1, and Optimize Wall Printing Order is enabled, then the order becomes:

1. R Wall 1.
2. R Wall 0.
3. H Wall 1.
4. H Wall 0.
5. H Skin Wall 0.
6. R Skin Wall 0.
7. H Skin Wall 1.
8. R Skin Wall 1.

Keeping the other constraints in place, I believe that could be optimised to:

1. R Wall 1.
2. R Wall 0.
3. H Wall 1.
4. H Wall 0.
5. H Skin Wall 0.
7. H Skin Wall 1.
6. R Skin Wall 0.
8. R Skin Wall 1.

Ghostkeeper, 2021-05-06:
Ah, that does appear to be fixed in Arachne.

With Arachne, we streamlined the Extra Walls for infill, skin and support to re-use the code used for the normal walls. Apparently that fixed the issue with multiple extra skin walls too.

I think there is nothing really to add any more then. This will be fixed by #1210.

richfelker, 2021-05-06:
> There is another constraint that you didn't take into account here, which is that each of the structures (walls, skin, etc.) must be printed in order. This is to reduce long-term flow changes, which [...]

I think this constraint should be optional if kept at all. It's irrelevant if you have a machine not sensitive to changes in speed, or if you have linear advance properly tuned, or if you print all structures at same speed. (All three apply for me.) Not having the excess travel back and forth would make a **huge** difference to overall print time, possibly cutting it in half. (Retract, travel to opposite wall, unretract often takes as long or longer than traversing the whole wall at speeds I print at, and that's even at 50 ms retraction.)

> Ah, that does appear to be fixed in Arachne.

Excellent! Is there any way to try out Arachne independently of the GUI? I use CuraEngine entirely from command line, and would like to see it, but can't find anything except the combined Cura application.

Ghostkeeper, 2021-05-17:
> Excellent! Is there any way to try out Arachne independently of the GUI? I use CuraEngine entirely from command line, and would like to see it, but can't find anything except the combined Cura application. Is it just the `libArachne_rebased` branch of this repo? (Edit: Seems like yes, I'm taking a look at how it works now.)

Indeed. As I understand, you're able to compile CuraEngine yourself. You'll just need the `libArachne_rebased` branch here then. In this branch, CuraEngine gets an extra dependency, the Boost header files, which it needs to calculate a Voronoi diagram. Let me know if it works. Otherwise I can also share my own compiled binary (or you could take the binary out of the AppImage of the Arachne Beta release).

richfelker, 2021-05-17:
Yes, it worked fine - that's the way I've been testing it. Thanks for confirming. It also needed the resource files from the corresponding branch of the Cura repo, but that was expected.

Ghostkeeper, 2021-12-14:
This is fixed now that https://github.com/Ultimaker/CuraEngine/pull/1210 is merged.

closed 2021-12-14 by Ghostkeeper