Ultimaker/CuraEngine #1443: Fold Z-hop into layer-change Z increment

TojNALvViV, 2021-04-22:
I noticed that currently, at the end of a layer, cura may decide to perform a Z-hop up (the first half of a hop), travel, and then start a new layer by performing a Z-hop down (the second half of the hop) immediately followed by a Z increment (to move to the new layer height.)  Note that my Z-hop height (0.1mm) is smaller than my layer height (0.25mm).

It would be a minor improvement to combine these last two Z movements in opposite directions into a single movement (in my case, a Z increment of 0.25 - 0.1 = 0.15mm)

**Application Version**
Arachne Engine beta

**Platform**
linux

fvrmr, 2021-05-07:
Hi @TojNALvViV thank you for your request. 
I will discuss this with the team. Keep you posted.

evtrados, 2021-05-17:
Developers, see issue CURA-8241.

TojNALvViV, 2021-05-22:
Thanks for the updates. :)

MRAAGH, 2022-03-18:
I also have this issue. I disable z-hop to reduce stringing. However, because layer change occurs before the last travel move (rather than after), stringing *still* occurs on one part of the model.