Ultimaker/CuraEngine #1446: speed_z_hop is used travel to next layer

richfelker, 2021-05-01:
In `LayerPlan::writeGCode`, `speed_z_hop` is used for the Z travel to the next layer, even though it may not even be a visible configuration item to the user. This causes unnecessary deceleration and acceleration at layer changes where the printer firmware (via junction deviation) could potentially just keep traveling at the same speed around two rapid 90° corners. The normal travel speed should just be used here, or it should be a separate configurable speed.

fvrmr, 2021-05-07:
Hi @richfelker the `speed z hop` is visible in the print settings. 
[image]

So you can change the speed here.

Let me know if that works for you.

fvrmr, 2021-05-07:
Then I misunderstood, I will discuss it with the team. 
Keep you posted!

Ghostkeeper, 2021-05-10:
We've decided to fix it in a different way, by making the Z Hop Speed visible if Z hops are not enabled.

Considering the default Z jerk of 0.4mm/s, it's going to slow down anyway for almost anyone. Only if the firmware source code is edited, recompiled and flashed with this explicit purpose will it be preventable to slow down for the layer change. The current Z hop speed is used to have a sensible and safe default.

Those who did edit their firmware source code may use a search-and-replace post-processing script to effectively implement a different speed for layer changes, if it's necessary for them to get a good print speed.

Ghostkeeper, 2021-05-13:
Indeed, for many things Cura assumes that the printer uses separate X/Y and Z jerk.