Ultimaker/CuraEngine #1448: Arachne regression: warts on outside of print

richfelker, 2021-05-07:
Using the experimental Arachne branch, I'm getting warts on the outside of a print that were not present with stable CuraEngine:

[image]

The model in question is hollow, in the actual source file not just printed with 0% infill, so that weight distribution is close to uniform (otherwise the thickness of a side along its normal would depend on its angle to the bed). Thus except at the very top and bottom it consists entirely of walls, no skin or infill, and it's rather thick (4-5 walls depending on angle).

I'm pretty sure what's happening is that the total extrusion for the affected layers is greater than the volume to extrude into. Since I'm printing outer walls first, the last inner wall has nowhere for all the filament to go, and it remains in the nozzle under pressure, causing the subsequent retraction to be effectively less than the configured (1mm) retraction. Then, when the nozzle moves to start the first outer wall of the next layer and unretracts, there's an excess of material right away that manifests as a wart on the outside of the print.

I've already set Maximum Extrusion Area Deviation to 0, in case it was the source of the problem, and it seemed to reduce the effect slightly but not entirely. I also played around a little with the `wall_transition_*` settings, but they didn't seem to make a difference.

My E-steps are pretty close to exact, slightly underextruding if anything according to measurement of filament moved, so I don't think it's just that Arachne is "too accurate". It might be that the toolhead's following of one inner wall path is cutting a corner too much, leaving a gap to the outer wall, and the extrusion in the next inner wall has no way to reach the gap to fill it. But even if it is some imprecision on my side, I suspect this will be an issue users will hit a lot with the new behavior when designing small solid parts consisting entirely of walls that need dimensional accuracy, and if my understanding of the problem is correct, there probably needs to be some way to mitigate it by printing the "last" wall sandwiched between walls with slightly less than 100% flow.

fvrmr, 2021-05-07:
Hi @richfelker 
You could try using `coasting`. I see that it is a dice and I am wondering if the numbers are on the outside or inside of the dice. 
If you want you could share your project file, then I can take a closer look.

richfelker, 2021-05-07:
Coasting could help confirm if this is the cause but it's not a solution. It would cause severe underextrusion of outer walls and other structures that aren't redundant.

richfelker, 2021-05-07:
I'm not using the frontend, only CuraEngine, so what would you like as the "project file"? Would the json config file I pass with `-j` work for you?

richfelker, 2021-05-11:
I've played with multiple perturbations of settings that might affect excess material in the walls, including even lowering inner wall flow by as much as 80%, with no major change. I'm going to see if I can find a simpler test case that doesn't require a 25-30 minute print for each trial and where I might be able to better analyze the moves in the affected layers.

richfelker, 2021-05-11:
Further update: problem disappears entirely with `wall_line_count=1` (and `infill_line_distance=0`) which produces just one outer wall and one "inner wall" (the outer wall of the inner surface). It's possible that's just perturbing something like the order of line printing and where travel/retraction occurs, but likely indicates there's just too much inner wall material being extruded somewhere, as I suspected. I might experiment by postprocessing the file sliced with original settings to just remove the inner walls (replacing them with a G92 to pretend the extrusion happened).

richfelker, 2021-05-11:
Looking at the output with an analyzer, the affected layers have something like 2.4-3.0 mm² more material extruded with Arachne than with stable, but less with the 80% extrusion modifier for inner walls where the problem still occurs. The warts seem to be where the previous layer has ended with extrusion in a constrained area where it should be less than full line width and the next (affected) layer starts at a visible point on the outer wall.

richfelker, 2021-05-11:
> Could it be that there were gaps in between the walls in 4.9 that allowed the material to flow in?

Yes, I think this is very likely. But it seems like either too much additional material is being extruded, or something about the path is preventing it all from being able to squeeze into the spots it's supposed to go.

I just tried with higher pressure advance (closer to calibrated instead of just lowest that works well) in case that would help, and it might have improved it a little bit, comparable to what the 80% inner-wall flow did, but didn't elliminate the problem.

Ghostkeeper, 2021-05-17:
Looking at the layer view of that slice
[image]
I think we might just be looking at the effect of the seam not being placed in the sharpest inner/outer corner. It's listed as a known issue here: https://github.com/Ultimaker/Cura/discussions/9388#discussion-3268674

richfelker, 2021-05-17:
The warts are where the seam would be, but normally I don't have any bulge from a seam, certainly nothing on the order of the nozzle width. So something else is going on.

I thought it might just be that Arachne is packing in the material so well that the retraction has nowhere to get air from, and siphons some material back into the nozzle that gets regurgitated at the unretract. I live-tweaked retraction speed to see if that would help, but it didn't. Also, turning on full combing (eliminating max combing distance without retract) so that it doesn't use retractions failed to make the issue go away entirely, although the warts became slightly smaller and smoother. So I still suspect there really is too much material being extruded at the end of the previous layer with not enough space for it to fit into.

richfelker, 2021-05-17:
I'm going to run some tests to see if it's just overextrusion exposed by Arachne's more thorough filling. Measured by filament length moved, my esteps are slightly underextruding (by ~1-2%) but my extruder pulls material very well and doesn't skip under backpressure and it's possible that that close to "nominal" level of extrusion is just too high on account of tiny gaps that form between lines/layers.

richfelker, 2021-05-17:
Based on testing with reduced flow for everything (at firmware level), I think it's plausible that this may just be overextrusion, where Arachne filling the space between the walls more completely triggered the outward symptom. It's possible to reduce the flow enough to make it negligible and doing so seems to improve other aspects of the prints anyway, or at least not harmful.

As reported, the symptom was happening with Arachne, and not with stable CuraEngine, so it's also plausible that Arachne is doing something wrong, possibly the known issue with wrong seam placement, making it more prominant.

At this time I don't see anything actionable except fixing the seam positioning deficiencies and re-testing with higher flow again after that's fixed, to see if it makes the problem go away. It might be nice to leave this open until then as a reminder to re-test.

Ghostkeeper, 2021-05-19:
> At this time I don't see anything actionable except fixing the seam positioning deficiencies and re-testing with higher flow again after that's fixed, to see if it makes the problem go away. It might be nice to leave this open until then as a reminder to re-test.

Agreed. We do expect that some profiles need to be re-optimised to achieve a similar result as before Arachne, and this seems to be such a situation. If it used to leave air gaps in between the walls and doesn't do that any more now, then previous compensations for underextrusion could now result in overextrusion. More strongly, there could be *effects* that were effectively compensating for underextrusion even if the user didn't intend them to.

We try to make the result reasonably similar by default, but it's such a drastic change that it's not going to be possible to ensure that no profiles need to be re-tuned.

We do have some CI tools in place that can be made to measure overlaps to see if the g-code itself theoretically overextrudes. Such things are part of our regression tests.

richfelker, 2021-07-05:
I think I have a better characterization of this issue now. Generally, it's a bad idea to print at 100% infill because, unless you're underextruding significantly, you'll be overextruding in some places due to minor imprecisions in ways that will cause the outer walls to bulge or for pressure to build up in the nozzle and produce warts. Pre-Arachne, it was very rare for thin parts to get printed as "100% infill" - it could only happen when they were exact multiples of the line width. Otherwise, the inner walls that wouldn't fit would get omitted (assuming some nonzero min flow, at least, and of course overlap compensation). But with Arachne, small details are basically guaranteed to get 100% infill. This makes them much more susceptible to dimensional inaccuracy and surface quality issues.

I think there should probably be some sort of functionality to mitigate this, possibly by leaving a gap or pulsing the flow of the innermost wall where opposing walls come in contact (without any infill region between them). Otherwise I expect print quality of parts with small components will diminish.

Ghostkeeper, 2021-07-12:
We have since implemented a feature for Arachne that prints the centremost wall last, so that the material can expand towards the centre: https://github.com/Ultimaker/CuraEngine/pull/1457

Perhaps that helps to reduce this issue. It was made because shine-through was observed and this was deemed to be one of the likely causes. Cura 4.10 and earlier will print the gap filling after the walls.

richfelker, 2021-07-12:
I'm already using `outer_inset_first`, so that change won't do anything. However, based on my reading of the linked PR (#1457), particularly this text:

> The odd-numbered centre walls are thought to have a problem with overextrusion, moreso than other walls (although why they overextrude is not yet known). This overextrusion propagates towards the outside, causing the surface to become more irregular.

suggests a mechanism for the bug I'm experiencing. In the absence of the ability to propagate outward (which would happen when printing inner walls first), the overextrusion propagates inward, until there's nowhere left to go. Then, when the last inner extrusion ends with a retraction, the excess material that had nowhere to go ends up just staying in the nozzle - the segment at the end that should be empty due to retraction - and comes out as a wart at the next unretract before the first extrude move starts.

richfelker, 2021-08-28:
After hitting a similar issue on another model today, I tried updating Arachne to latest. That greatly improved print quality and entirely fixed today's occurrance. After that I reprinted the D20 test and got much better results than before, but still not perfect. The remaining issue may be different, however; it seems to be doing some strange travel moves outside the model rather than keeping the nozzle inside. I'll try to get a better report on what's going on soon, but I think it's likely that whatever was causing this bug may have been fixed now. Let's keep it open until I can better confirm that.

richfelker, 2021-08-28:
I would not be surprised if 82245250763e230c001e706b3b2c9727fd0b55e5 was the commit that fixed this.

richfelker, 2021-09-08:
> We still have a known issue with combing open for Arachne. That could be what you're seeing.

Is there a tracker item I could add information on if I have anything relevant? The odd travels outside the object seem to be happening in layers where the full region is connected but, with the outer walls removed, becomes disconnected. Rather than comb to the spot nearest to the destination inside the inner-walls component and move across the single outer wall between them, Arachne immediately crosses the outer wall, performs avoiding travel outside the model, then again crosses an outer wall to get back inside where it wants to start printing again. This results in potentially marring two more-visible outer wall points rather than one outer-wall point in a (naturally, due to the concavity leading to this behavior) less-visible location.

If it matters, for my configuration the retractions would be the same either way, because I set max combing distance without retract to 0.8 mm. But in the absence of retraction while combing, I think the path I'd prefer it take is even more preferable since it could avoid retraction too.

Ghostkeeper, 2021-09-17:
The tracking link we have is this: https://github.com/Ultimaker/Cura/discussions/9388#discussioncomment-593924

richfelker, 2022-02-22:
Now that #1562 is merged, I'll give this another try and see if anything else has changed.

richfelker, 2022-02-23:
The failure to comb (travel outside the model) still seems to be happening, but otherwise results are looking very good.