Ultimaker/CuraEngine #1452: Maximum Travel Resolution default significantly slows movement

richfelker, 2021-05-28:
The maximum travel resolution default of 1 mm causes travel around finely-approximated curved paths to become very coarse, replacing many slight turns with fewer turns at greater angle. On firmware that uses the junction deviation algorithm for maximum velocity through a non-straight junction (modern Marlin versions, Klipper, etc.), these greater angles impose a much lower maximum junction velocity, thereby making the travel slow. For reference, note that if 180-theta is the exterior angle (theta the interior angle) at the junction, the junction velocity limit scales as sqrt(sin(theta/2)/(1-sin(theta/2))).

There does not seem to be any good technical reason for the travel default to be higher than the corresponding default for print moves, but just making them match (0.5) wouldn't help anyway. The values don't translate directly since the deviation setting, not the resolution setting, is usually the dominant factor for print moves, and there is no maximum deviation setting for travel. I'm just using a much lower setting as a workaround, but more thought is probably needed on the correct fix.

fvrmr, 2021-06-03:
Hi @richfelker lowering the settings is the way it should work because this is logical to the printer.
The lower the value the travel moves will have less smooth corners.

Ghostkeeper, 2021-06-07:
The default for the Maximum Travel Resolution is higher than the Maximum Resolution by ratio of the travel speed to the print speed.

This assumes that the Maximum Resolution is chosen such that the printer doesn't run into buffer underruns at the normal print speed. The Maximum Travel Resolution is then calculated such that travel moves also don't run into buffer underruns at their increased speed. Yes, the Maximum Resolution is also limited by the Maximum Deviation in some cases, but not in other cases, depending on the radius of the corner.

Reducing the Maximum Travel Resolution setting is the only way to get smoother corners in your travel moves then.

If you can reduce the Maximum Travel Resolution setting without getting buffer underruns, then by all means do so. Although most likely you can also reduce the Maximum Resolution setting then, in theory. We're not likely to change the default here, but you're of course free to change the setting for yourself.

Is there any change you would like to propose, other than changing the default?