Ultimaker/CuraEngine #1453: [Question] Is "print_sequence" "one_at_a_time" still supported?

marciot, 2021-06-04:
I am trying to understand why this parameter which is listed in "fdmprinter.json" is having no effect and grepping through the source code for CuraEngine I was unable to find either "print_sequence", "one_at_a_time" or "all_at_once" anywhere. I assume that if these strings aren't present then Cura Engine certainly can't act on them. Has this parameter been deprecated? Or am I just missing them?

nallath, 2021-06-07:
It's a setting that is only used by the front end. Basicly we change the way that the objects are sent to the engine. You can see how it's different here: https://github.com/Ultimaker/Cura/blob/78ecb7ec9a0c1deae6c322a286a13ee70a07e379/cura/OneAtATimeIterator.py

closed 2021-06-07 by nallath

marciot, 2021-06-07:
@nallath: Thank you for clarifying, this is what I had begun to suspect. So, looking through that code, it looks like most of it pertains to validating whether the models are non-overlapping and validating that it can be printed at one time. How are those objects eventually sliced? Is the Cura Engine called multiple times, with one model per call, and then the resulting GCODE merged together in post-processing, or is there are way to pass multiple items to Cura Engine while having it slice the objects sequentially? If the latter, is this interface exposed via the command line or only via the Arcus interface?

Ghostkeeper, 2021-06-14:
The models are sent in individual "mesh groups". A "mesh group" is a term internal to CuraEngine (and admittedly a bit weird) denoting a group of meshes that gets printed from bottom to top. [(Documentation here.)](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#printing-order)

CuraEngine does create the g-code for all mesh groups in one call. It iterates over the mesh groups here:

https://github.com/Ultimaker/CuraEngine/blob/aa69bb230031da98d64d024f328ac1c55acdf57c/src/Slice.cpp#L18

Some things, such as the start g-code, are only added once.