Ultimaker/CuraEngine #1456: brew install CuraEngine - can't find or open JSON file

NGK98, 2021-06-10:
My team and I are mechanical engineers and are not software strong, but need to build an automation for our 3D printing workflow. I am trying to get CuraEngine set up on my mac (MacOS Catalina, version: 10.15.7). I used brew install CuraEngine (trying to get around a libArcus issue I am having https://github.com/Ultimaker/libArcus/issues/118). When running the CuraEngine slice command the following output happens where I am getting errors about finding and opening the JSON file but then it looks like it is still able to produce the gcode.. Can someone tell me if this is working??

**Terminal**
Natalie@Nats-MBP Cura % CuraEngine slice -v -j resources/definitions/ultimaker2_extended.def.json -l ~/Desktop/Scan1.stl -o ~/Desktop/outputGcode1.gcode

Cura_SteamEngine version master
Copyright (C) 2020 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
[WARNING] JSON setting material_print_temperature has no default_value!
[WARNING] JSON setting material_bed_temperature has no default_value!
[WARNING] JSON setting material_bed_temperature_layer_0 has no default_value!
[WARNING] JSON setting material_standby_temperature has no default_value!
[WARNING] JSON setting extruder_prime_pos_y has no default_value!
[WARNING] JSON setting extruder_prime_pos_x has no default_value!
[WARNING] JSON setting relative_extrusion has no default_value!
[ERROR] Couldn't find definition file with ID: ultimaker2_extruder_0
[ERROR] Couldn't open JSON file: 
[WARNING] JSON setting machine_start_gcode has no default_value!
[WARNING] JSON setting machine_end_gcode has no default_value!
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
[WARNING] JSON setting gantry_height has no default_value!
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[ERROR] Couldn't find definition file with ID: ultimaker2_extended_extruder_0
[ERROR] Couldn't open JSON file: 
[WARNING] Mesh has disconnected faces!
loading '/Users/user/Desktop/Scan1.stl' took 0.038 seconds
Loaded from disk in 0.062s
[WARNING]  -s mesh_position_y="0" -s center_object="false" -s small_feature_speed_factor_0="50" -s small_feature_speed_factor="50" -s wipe_hop_speed="10" -s wipe_pause="0" -s wipe_retraction_prime_speed="2" -s wipe_retraction_extra_prime_amount="0" -s wipe_retraction_enable="true" -s clean_between_layers="false" -s bridge_skin_density_3="80" -s bridge_skin_density_2="75" -s bridge_skin_material_flow_2="100" -s bridge_skin_speed_2="25" -s bridge_enable_more_layers="true" -s bridge_skin_density="100" -s bridge_skin_material_flow="60" -s bridge_wall_material_flow="50" -s bridge_wall_coast="100" -s bridge_sparse_infill_max_density="0" -s bridge_skin_support_threshold="50" -s bridge_settings_enabled="false" -s adaptive_layer_height_threshold="0.2" -s wireframe_straight_before_down="20" -s wireframe_drag_along="0.6" -s wireframe_top_jump="0.6" -s wireframe_up_half_speed="0.3" -s wireframe_flat_delay="0.1" -s wireframe_bottom_delay="0" -s wireframe_flow_connection="100" -s wireframe_printspeed_up="5" -s wireframe_printspeed_bottom="5" -s wireframe_enabled="false" -s magic_fuzzy_skin_enabled="false" -s support_conical_min_width="5" -s support_conical_angle="30" -s adaptive_layer_height_variation="0.1" -s cross_infill_density_image="" -s coasting_speed="90" -s coasting_min_volume="0.8" -s coasting_enable="false" -s conical_overhang_hole_size="0" -s conical_overhang_enabled="false" -s wireframe_printspeed_flat="5" -s draft_shield_height_limitation="full" -s draft_shield_dist="10" -s draft_shield_enabled="false" -s support_zag_skip_count="5" -s support_skip_some_zags="false" -s minimum_polygon_circumference="1" -s material_flow_temp_graph="[[3.5,200],[7.0,240]]" -s infill_enable_travel_optimization="false" -s roofing_angles="[ ]" -s roofing_pattern="lines" -s roofing_line_width="0.4" -s slicing_tolerance="middle" -s smooth_spiralized_contours="true" -s magic_spiralize="false" -s magic_mesh_surface_mode="normal" -s mold_angle="40" -s mold_roof_height="0.5" -s layer_height="0.1" -s material_break_preparation_retracted_position="-16" -s infill_mesh="false" -s support_extruder_nr_layer_0="0" -s meshfix_maximum_travel_resolution="1" -s flow_rate_extrusion_offset_factor="100" -s acceleration_wall_x="3000" -s carve_multiple_volumes="true" -s switch_extruder_extra_prime_amount="0" -s machine_nozzle_temp_enabled="true" -s material_maximum_park_duration="300" -s skirt_brim_speed="30" -s switch_extruder_retraction_amount="20" -s ooze_shield_dist="2" -s machine_endstop_positive_direction_x="false" -s ooze_shield_angle="60" -s support_line_width="0.4" -s ooze_shield_enabled="false" -s mesh_position_x="0" -s prime_tower_position_y="200" -s bridge_wall_min_length="5" -s prime_tower_position_x="200" -s raft_surface_acceleration="3000" -s raft_base_line_width="0.8" -s machine_nozzle_heat_up_speed="2" -s material_flush_purge_length="60" -s raft_interface_line_spacing="0.9" -s raft_interface_thickness="0.15" -s speed_topbottom="30" -s raft_smoothing="5" -s brim_replaces_support="true" -s print_sequence="all_at_once" -s acceleration_support_infill="3000" -s brim_gap="0" -s raft_surface_layers="2" -s skirt_brim_minimal_length="250" -s mold_width="5" -s adhesion_extruder_nr="0" -s extruder_prime_pos_y="0" -s meshfix_keep_open_polygons="false" -s prime_blob_enable="false" -s support_mesh_drop_down="true" -s extruder_prime_pos_z="0" -s support_tower_roof_angle="65" -s support_tower_diameter="3" -s support_use_towers="true" -s minimum_roof_area="1" -s jerk_topbottom="20" -s remove_empty_first_layers="true" -s prime_tower_size="20" -s support_mesh="false" -s support_roof_pattern="concentric" -s support_bottom_line_distance="0.4" -s support_interface_skip_height="0.3" -s support_bottom_enable="false" -s minimum_support_area="0" -s support_roof_height="1" -s material_bed_temperature_layer_0="60" -s support_infill_sparse_thickness="0.1" -s wireframe_fall_down="0.5" -s support_offset="0" -s wipe_hop_amount="1" -s support_join_distance="2" -s support_bottom_stair_step_min_slope="10" -s support_bottom_stair_step_width="5" -s support_xy_distance_overhang="0.2" -s machine_nozzle_tip_outer_diameter="1" -s min_infill_area="0" -s support_xy_overrides_z="z_overrides_xy" -s support_xy_distance="0.7" -s machine_steps_per_mm_e="1600" -s material_crystallinity="false" -s support_bottom_distance="0.1" -s support_brim_enable="false" -s support_infill_angles="[ ]" -s raft_base_fan_speed="0" -s support_tree_collision_resolution="0.4" -s support_tree_branch_diameter_angle="5" -s machine_steps_per_mm_y="50" -s support_tree_branch_diameter="2" -s support_tree_branch_distance="1" -s machine_max_jerk_z="0.4" -s support_tree_angle="40" -s support_structure="normal" -s support_enable="false" -s cool_min_layer_time="5" -s cool_min_layer_time_fan_speed_max="10" -s meshfix_maximum_deviation="0.025" -s wall_overhang_angle="90" -s cool_fan_enabled="true" -s cool_fan_speed_max="100" -s wipe_move_distance="20" -s cool_fan_speed_min="100" -s support_bottom_extruder_nr="0" -s jerk_wall_0="20" -s retraction_hop="1" -s material_print_temp_wait="true" -s support_roof_angles="[ ]" -s switch_extruder_retraction_speed="20" -s acceleration_roofing="3000" -s retraction_hop_enabled="false" -s wall_x_extruder_nr="-1" -s switch_extruder_prime_speed="20" -s build_volume_temperature="0" -s extruder_prime_pos_x="0" -s layer_start_y="0" -s machine_min_cool_heat_time_window="50" -s layer_start_x="0" -s raft_interface_speed="15" -s support_meshes_present="false" -s travel_avoid_distance="0.625" -s acceleration_infill="3000" -s travel_avoid_supports="false" -s raft_surface_speed="20" -s machine_height="305" -s travel_retract_before_out [... cut ...]
Progress: inset+skin accomplished in 7.785s
Starting support...
Processing platform adhesion
Progress: support accomplished in 1.292s
Starting export...
Progress: export accomplished in 13.583s
Total time elapsed 22.95s.
Gcode header after slicing:
;FLAVOR:UltiGCode
;TIME:100390
;MATERIAL:207745
;MATERIAL2:0
;NOZZLE_DIAMETER:0.4
;MINX:63.692
;MINY:60.138
;MINZ:0.3
;MAXX:159.045
;MAXY:170.137
;MAXZ:104.9
End of gcode header.
Print time (s): 100390
Print time (hr|min|s): 27h 53m 10s
Filament (mm^3): 207745

Ghostkeeper, 2021-06-18:
The warnings here are to be expected. CuraEngine doesn't know about all of the data types that Cura uses for its settings. In particular, Cura's front end needs to use a couple of settings to determine the disallowed areas. Those settings are lists of coordinates and CuraEngine doesn't understand them. CuraEngine also doesn't need to use them though (which is why we never implemented parsing those) so it's all good.

A bit more of a concern is that CuraEngine is not able to parse the `"value"` of these settings. Those are Python expressions and CuraEngine has no Python interpreter built-in.  It only uses `"default_value"`. It is warning you that you have a setting override in `ultimaker2.def.json` which is useless because it has no `"default_value"`. It'll continue to slice because it uses the default from `fdmprinter.def.json` then, but your slice result won't have the correct setting value.

It also can't find the `ultimaker2_extruder_0.def.json` file. It only looks in the same folder as the provided definition and in the working directory, so if the resource structure of Cura's front-end is left intact it will indeed not find this file. Instead it will use `fdmextruder.def.json` which is in the `definitions` folder. In this case this won't do you any harm though since `ultimaker2_extruder_0.def.json` doesn't seem to change any settings away from the defaults. But for other printers this could be problematic.

So in brief, it slices because it falls back to the `default_value`s in fdmprinter.def.json. There's no real need to be concerned, since your settings won't be perfect for the Ultimaker 2 anyway without also applying the profiles for the nozzle, material and quality you want to slice for. But it is something you might want to look at anyway, depending on your application.

closed 2021-06-18 by Ghostkeeper

siamak06, 2025-03-18:
https://github.com/Ultimaker/Cura/tree/main/resources/definitions