Ultimaker/CuraEngine #1458: Feature request: unretract inside region before printing outer wall

richfelker, 2021-06-25:
When starting an outer wall - particularly, with `outer_inset_first`, and particularly on an overhang - the unretract can cause warts on the outer surface. Provided the region is more than the nozzle width thick at the starting point, a point inset just slightly inside the outer wall (preferably by a configurable amount, default to half nozzle width or so) could be used to perform the unretract, followed by travel to the real starting point to start extrusion. This is sort of an "inverse wipe".

Ghostkeeper, 2021-07-05:
It's expected to already do this. That's what this little hook is for when travelling to the outer wall.
[image]

However when I test this, it appears like it's unretracting too late. It's supposed to be unretracting while away from the outer wall, but it seems like that unretract happens one move later when it actually starts to print.

richfelker, 2021-07-05:
So the intended behavior is what I described, but there seems to be a bug preventing it from actually doing what was intended here?

Ghostkeeper, 2021-07-06:
Indeed. When moving to print the outer wall, it first finds a spot next to the outer wall where it can unretract. It should move there instead, unretract, and then move to the start of the outer wall. From what I can see it currently does move there, then move to the outer wall, and then unretract, which is wrong. Probably the code to unretract is missing and then automatically filled in when it actually starts printing the outer wall.

The spot near the outer wall does have a limit for distance though, I think it's like 2mm. So it won't function for thin walls (where everything is outer wall) and very sharp tips.

Ghostkeeper, 2021-07-06:
Developers, see issue CURA-8365.

richfelker, 2021-07-06:
> The spot near the outer wall does have a limit for distance though, I think it's like 2mm

Do you mean there needs to be space to move back 2mm into the shape? That's large enough that it wouldn't work for most models I print. 1-1.5 nozzle widths should be plenty, and even half a nozzle width should be able to achieve the desired goal most of the time. 2mm sounds like way too much; it would allow oozing between the unretract and the wall start, yielding a weak wall.

I think the ideal behavior should be tunable, by default trying to go in 1 nozzle width, but going less (e.g. just half that if the entire region is 2 outer walls butted against each other) if the default would cross an opposing outer wall. This is probably computed by insets, so it would be something like "find a point within distance (default: nozzle width) in the 1 nozzle width inset, repeat with reduced inset if none exist".

richfelker, 2021-07-06:
Hm from the image you posted it looks like it's just doing a 1 line width inset, so perhaps I misunderstood and the 2mm is a search range. Still I'm not clear on what happens if there's a start point with nowhere inside to unretract adjacent to the start point, but one 2 mm away. Does it (after fixing the missing unretract) then move the start point, or do 2 mm of unretracted travel?

richfelker, 2021-07-06:
OK, I found the relevant code in `LayerPlan::addTravel` and `moveInsideCombBoundary`, but I don't see any way to represent the required unretract operation in the path they emit. Unretractions are only generated (along with additional priming, if requested) as part of emitting the extrusion move via `GCodeExport::writeExtrusion`. So I don't see an easy fix.

Ghostkeeper, 2021-07-13:
> Do you mean there needs to be space to move back 2mm into the shape?

No, I meant that it attempts to move away from the outer wall, unless that would cause a travel move longer than 2mm. With most models, like the cube above, the distance to get away from the outer wall is less than 2mm. Usually 0.2mm (with 0.4mm line width) or a bit more in the corner. But if the seam is placed in a very thin part or a very sharp corner, the distance to be away from the outer wall can be much longer. Like with this example, it would need to travel along the entire length of the fan blade to get away from an outer wall, if the seam is at the tip of the blade:
[image]

But that 2mm is just from memory. I don't really remember what the actual distance was.

richfelker, 2021-08-29:
From the linked PR, it looks like this should be fixed now. Is that the case?

richfelker, 2021-08-29:
OK, based on a rebuild from latest git, the bug appears to be fixed. I see the unretracts happening before the final short move to the outer wall start position, rather than after it.

Ghostkeeper, 2021-09-08:
Yes, that PR was intended to fix the issue.
Thanks for the help!

closed 2021-09-08 by Ghostkeeper