Ultimaker/CuraEngine #1463: Feature request: split walls/skins for bridging

richfelker, 2021-07-04:
At least as I understand it, CuraEngine presently applies bridge settings to an entire line or skin region that may be partly bridging and partly supported, based on the percentage of length/area that's unsupported. This inherently causes wrong settings to be used for one or the other. Instead, when the bridge length (for walls) or area (for skins) exceeds some threshold, the line or region should be split into bridge and non-bridge parts so that the appropriate settings can be applied.

This is particularly important when using reduced flow for bridges to prevent sagging; if the reduced flow is also used for the non-bridged parts, the connected part of the layer will fail to adhere, making the part weak adjacent to the bridge. It's also important if postprocessing the output to apply different Linear Advance factor to bridges (which is needed for correct extrusion, since the factor normally used is not correct when there's no backpressure from extruding against an existing surface).

smartavionics, 2021-07-05:
My Cura build does partition skin regions depending on whether they are above air or not. It does this so that it can optimise the direction of the skin lines depending on the shape of the bridge region. A side effect of this partitioning is that the portions of the skin that are supported will be printed using the normal skin settings rather than the bridge skin settings. Hope this helps.

richfelker, 2021-07-05:
That's good to know. Does it do it for bridge walls too? Those matter a lot more in some sense for under/over extrusion of the non-bridge part. In any case though I'd like to see this in upstream CuraEngine since I'm trying to follow it especially with the move to Arachne.

smartavionics, 2021-07-05:
> Does it do it for bridge walls too? Those matter a lot more in some sense for under/over extrusion of the non-bridge part. 

I don't understand your question as UM Cura already splits walls into bridge and non-bridge sections and uses different values for speed and flow when printing the bridge sections. Isn't that what you are asking for?

richfelker, 2021-07-05:
Are you sure it does? From past experience I'm pretty sure it doesn't, and the threshold setting descriptions seem to imply it's not intended to. If needed I can make some test cases.

Ghostkeeper, 2021-07-12:
Indeed, it adjusts the part of the wall that is bridging, as well as a small distance before and after the bridge as determined by the Bridge Wall Coasting setting.
[image]

This does not happen for skin. The skin area is still handled as a single area, with its parameters adjusted according to the bridge settings. It is not split up. This was how it was in the PR as the feature was implemented, and never got the attention it deserves. It should really be considered a requirement if we are to bring the bridging feature out of the experimental stage.

Ghostkeeper, 2021-07-15:
Developers, see issue CURA-8391.